zephyrproject-rtos/zephyr #74364: net: ptp: maybe unused clock variable in global scope causing issues

cfriedt, 2024-06-16:
**Describe the bug**

I was getting a build failure due to a variable `clock` being defined in the global scope (although statically) because it conflicts with the ISO C89 `clock()` function. 

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using? `nucleo_h563zi`
 - What have you tried to diagnose or workaround this issue? Created a PR
 - Is this a regression? 74a25772c55c77b09d0b6b1aaced850610a77ed4

**To Reproduce**
Steps to reproduce the behavior:
1. `twister -i -p nucleo_h563zi -s samples/net/ptp/sample.net.ptp`
2. See error

**Expected behavior**
The build should succeed

**Impact**
Showstopper for ZVFS integration

**Logs and console output**
```shell
nucleo_h563zi             samples/net/ptp/sample.net.ptp                      ERROR Build failure (build)
...
subsys/net/lib/ptp/clock.c:58:25: error: 'clock' redeclared as different kind of symbol
   58 | static struct ptp_clock clock = { 0 };
...
/opt/toolchains/zephyr-sdk-0.16.8/arm-zephyr-eabi/picolibc/ include/time.h:100:12: note: previous declaration of 'clock' with type 'clock_t(void)' {aka 'long unsigned int(void)'}
  100 | clock_t    clock (void);
...
subsys/net/lib/ptp/clock.c:58:25: error: 'clock' defined but not used [-Werror=unused-variable]
   58 | static struct ptp_clock clock = { 0 };
```

**Environment (please complete the following information):**
 - OS: CI / Linux
 - Toolchain: Zephyr SDK 0.16.8
 - Commit: 74a25772c55c77b09d0b6b1aaced850610a77ed4

**Additional context**

rlubos, 2024-06-17:
FYI @awojasinski, for some reason I cannot assign the issue to you.

awojasinski, 2024-06-17:
@rlubos probably I need to be added to github maintainers. I'll take care for that. Thanks for letting me know

awojasinski, 2024-06-17:
The https://github.com/zephyrproject-rtos/zephyr/pull/74366 looks good. I approved it.

PR #74366 (2024-06-18): net: ptp: do not use a reserved id in global scope
This fixes an issue where the statically scoped variable "clock" in the global context causes an issue with CI.

Fixes #74364

Cherry-picked from https://github.com/zephyrproject-rtos/zephyr/pull/73978

Generally, we should avoid using reserved names in global contexts. 

```shell
subsys/net/lib/ptp/clock.c:58:25: \
  error: 'clock' redeclared as different kind of symbol
   58 | static struct ptp_clock clock = { 0 };

/opt/toolchains/zephyr-sdk-0.16.8/arm-zephyr-eabi/picolibc/ \
  include/time.h:100:12: \
  note: previous declaration of 'clock [... cut]
closed 2024-06-18 by nashif