zephyrproject-rtos/zephyr #74367: Waveshare st7789v2 with 240x280 screen shows a line at the bottom of the display

ljunquera, 2024-06-16:
I purchased a new Waveshare ST7789 display, but was getting a line at the bottom of the screen and cannot determine why. 

[image]

The display is a 240x280:
https://www.waveshare.com/wiki/1.69inch_Touch_LCD_Module

I am using the driver:
https://github.com/zephyrproject-rtos/zephyr/blob/main//boards/shields/st7789v_generic/doc/index.rst

And nRF Connect for VS Code and changed the height to 280:

```
       st7789v_st7789v_tl019fqv01: st7789v@0 {
        compatible = "sitronix,st7789v";
        spi-max-frequency = <20000000>;
        reg = <0>;
        cmd-data-gpios = < &gpio1 11 GPIO_ACTIVE_LOW>;
        reset-gpios = < &gpio1 10 GPIO_ACTIVE_LOW>;
        width = <240>;
        height = <280>;
        x-offset = <0>;
        y-offset = <0>;
        vcom = <0x19>;
        gctrl = <0x35>;
        vrhs = <0x12>;
        vdvs = <0x20>;
        mdac = <0x00>;
        gamma = <0x01>;
        colmod = <0x05>;
        lcm = <0x2c>;
        porch-param = [0c 0c 00 33 33];
        cmd2en-param = [5a 69 02 01];
        pwctrl1-param = [a4 a1];
        pvgam-param = [D0 04 0D 11 13 2B 3F 54 4C 18 0D 0B 1F 23];
        nvgam-param = [D0 04 0C 11 13 2C 3F 44 51 2F 1F 1F 20 23];
        ram-param = [00 F0];
        rgb-param = [CD 08 14];
	};
```

aescolar, 2024-06-18:
@ljunquera could you please provide more information regarding which Zephyr version you are using

ljunquera, 2024-06-18:
Zephyr v3.5.0

ljunquera, 2024-06-21:
That did it! Thank you.

```
    st7789v_st7789v_tl019fqv01: st7789v@0 {
        compatible = "sitronix,st7789v";
        spi-max-frequency = <20000000>;
        reg = <0>;
        cmd-data-gpios = < &gpio1 11 GPIO_ACTIVE_LOW>;
        reset-gpios = < &gpio1 10 GPIO_ACTIVE_LOW>;
        width = <240>;
        height = <280>;
        x-offset = <0>;
        y-offset = <20>;
        vcom = <0x19>;
        gctrl = <0x35>;
        vrhs = <0x12>;
        vdvs = <0x20>;
        mdac = <0x00>;
        gamma = <0x01>;
        colmod = <0x05>;
        lcm = <0x2c>;
        porch-param = [0c 0c 00 33 33];
        cmd2en-param = [5a 69 02 01];
        pwctrl1-param = [a4 a1];
        pvgam-param = [D0 04 0D 11 13 2B 3F 54 4C 18 0D 0B 1F 23];
        nvgam-param = [D0 04 0C 11 13 2C 3F 44 51 2F 1F 1F 20 23];
        ram-param = [00 F0];
        rgb-param = [CD 08 14];
    };
```

closed 2024-06-21 by ljunquera