zephyrproject-rtos/zephyr #74369: tests:  lpcxpresso55s69/lpc55s69/cpu0/ns: drivers.flash.common.tfm_ns: Build failure

hakehuang, 2024-06-16:
**Describe the bug**
tests/drivers/flash/common build failure

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 lpcxpresso55s69/lpc55s69/cpu0/ns
 - What have you tried to diagnose or workaround this issue?
 git bisect
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
 

```
eccc2ba8c47591e8ab0fced054a2f3e26e2360e3 is the first bad commit
commit eccc2ba8c47591e8ab0fced054a2f3e26e2360e3
Author: Swift Tian <swift.tian@ambiq.com>
Date:   Tue Apr 23 15:47:29 2024 +0800

    drivers: flash: Add a flash emulator
    
    Add a flash emulator that uses MSPI API to emulate a flash device
    on MSPI bus.
    
    Signed-off-by: Swift Tian <swift.tian@ambiq.com>

 drivers/flash/CMakeLists.txt           |   1 +
 drivers/flash/Kconfig                  |   2 +
 drivers/flash/Kconfig.mspi             |  24 ++
 drivers/flash/flash_mspi_emul_device.c | 458 +++++++++++++++++++++++++++++++++
 4 files changed, 485 insertions(+)
 create mode 100644 drivers/flash/Kconfig.mspi
 create mode 100644 drivers/flash/flash_mspi_emul_device.c
```

**To Reproduce**
Steps to reproduce the behavior:

scripts/twister -p lpcxpresso55s69/lpc55s69/cpu0/ns -T tests/drivers/flash/common -s drivers.flash.common.tfm_ns

**Expected behavior**
Build PASS

**Impact**

flash support for this platform

**Logs and console output**

```
CMakeFiles/app.dir/src/main.c.obj -c /home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/tests/drivers/flash/common/src/main.c
/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/tests/drivers/flash/common/src/main.c:49:2: error: #error There is no flash device enabled or it is missing Kconfig options
   49 | #error There is no flash device enabled or it is missing Kconfig options
      |  ^~~~~
In file included from /home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/include/zephyr/toolchain/gcc.h:98,
                 from /home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/include/zephyr/toolchain.h:50,
                 from /home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/include/zephyr/kernel_includes.h:23,
                 from /home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/include/zephyr/kernel.h:17,
                 from /home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/tests/drivers/flash/common/src/main.c:7:
/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/include/zephyr/device.h:92:41: error: '__device_dts_ord_65' undeclared here (not in a function); did you mean '__device_dts_ord_15'?
   92 | #define DEVICE_NAME_GET(dev_id) _CONCAT(__device_, dev_id)
      |                                         ^~~~~~~~~
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used:  v3.7.0-rc1

dleach02, 2024-06-25:
@hakehuang something changed in your test environment as this test case never compiled for cpu0/ns or for cpu1 ? I went back to v3.6.0 and it failed there. The problem is that the storage node in the flash seems to only be enabled for cpu0.

hakehuang, 2024-06-25:
> @hakehuang something changed in your test environment as this test case never compiled for cpu0/ns or for cpu1 ? I went back to v3.6.0 and it failed there. The problem is that the storage node in the flash seems to only be enabled for cpu0.
the test environment is not changed, this case is enabled for a long time. and according to my log it builds ok(v3.6.0-4291-gd7dd64) before v3.7.0.  it is a build only test.
```
  drivers.flash.common.tfm_ns:
    build_only: true
    filter: (CONFIG_FLASH_HAS_DRIVER_ENABLED and CONFIG_TRUSTED_EXECUTION_NONSECURE
      and dt_label_with_parent_compat_enabled("storage_partition", "fixed-partitions"))
    integration_platforms:
      - nrf9161dk/nrf9161/ns
```

butok, 2024-06-25:
> @butok, what is the expected configuration/behavior if you are doing _ns?

- **_ns**: assumes using  TFM. The ROM API is placed (by default) in the secure area, so the iap will not work for **ns**.
- **cpu1**: never tested. Guess, it should support IAP.

closed 2024-06-25 by hakehuang