zephyrproject-rtos/zephyr #74375: SWO logging fails to compile for STM32H7

LnnrtS, 2024-06-16:
```
west build -p always -b stm32h735g_disco samples/hello_world -DCONFIG_LOG=y -DCONFIG_LOG_PRINTK=y -DCONFIG_LOG_BACKEND_SWO=y`
```

fails with `error: 'DBGMCU_CR_TRACE_IOEN' undeclared ` in `soc/st/stm32/common/soc_config.c` 

Using Zephyr SDK `0.16.8`

Probably similar to #44584

FRASTM, 2024-06-18:
Yes, the stm32H7 serie has no LL_DBGMCU_SetTracePinAssignment function 
and no DBGMCU_CR_TRACE_IO enable bit in its DBGMCU_CR register

FRASTM, 2024-06-18:
this kind of patch should fix for all stm32 serie:
```patch
diff --git a/soc/st/stm32/common/soc_config.c b/soc/st/stm32/common/soc_config.c
index 70f270cf494..36d87838f70 100644
--- a/soc/st/stm32/common/soc_config.c
+++ b/soc/st/stm32/common/soc_config.c
@@ -30,9 +30,9 @@ static int st_stm32_common_config(void)
 #if defined(CONFIG_SOC_SERIES_STM32WBX) || defined(CONFIG_SOC_SERIES_STM32H5X)
 	LL_DBGMCU_EnableTraceClock();
 #endif
-#if !defined(CONFIG_SOC_SERIES_STM32WBX)
+#ifdef DBGMCU_CR_TRACE_IOEN
 	LL_DBGMCU_SetTracePinAssignment(LL_DBGMCU_TRACE_ASYNCH);
-#endif
+#endif /* DBGMCU_CR_TRACE_IOEN */
 #endif /* CONFIG_LOG_BACKEND_SWO */
 
 #if defined(CONFIG_USE_SEGGER_RTT)
```

closed 2024-06-18 by aescolar