zephyrproject-rtos/zephyr #74377: DNS Resolver Double Free

EricNRS, 2024-06-16:
**Describe the bug**
Seeing double free errors related to the DNS dispatcher on ESP32-S3 target.

```
uart:~$ net dns query google.com
Query for 'google.com' sent.
dns: 142.250.67.14
dns: All results received
[00:00:32.258,000] <err> net_buf: recv_data():148: buf 0x3fcb1894 double free
```

**To Reproduce**
To reproduce using WiFi test application:
`west build --pristine=auto -b xiao_esp32s3/esp32s3/procpu samples/net/wifi -DCONFIG_DNS_RESOLVER=y && west flash && west espressif monitor`

```
*** Booting Zephyr OS build v3.7.0-rc1-8-g4d0a8c1e8b8 ***
uart:~$ wifi connect -s "SSID" -p PASSWORD -k 1
Connection requested
Connected
[00:00:19.884,000] <inf> net_dhcpv4: Received: 192.168.1.45
uart:~$ net dns query google.com
Query for 'google.com' sent.
dns: 142.250.67.14
dns: All results received
[00:00:32.258,000] <err> net_buf: recv_data():148: buf 0x3fcb1894 double free
uart:~$ 
```

**Impact**
Unknown impact -- appears to just be a nuisance error.

**Environment (please complete the following information):**
Zephyr v3.7.0-rc1 at commit 4d0a8c1e8b8

EricNRS, 2024-06-18:
I am getting a large amount of errors in my full application along with crashes every few minutes during stress tests.  One crash pointed to packet cursor which would indicate use-after-free, but I didn't capture it :(.

I am out of time to look at it further today, but have disabled DNS to run a stress test overnight to confirm it isn't some other issue and so far it has run for almost an hour when the longest run I got previously was about 12 minutes.

I had a quick look through the code in dispatcher.c and didn't see anything, but haven't looked through the callbacks.

@jukkar - have you had a chance to look at this and is use-after free possible?

```
[00:00:48.614,000] <err> net_dns_dispatcher: DNS recv error (-2)
[00:00:48.717,000] <err> net_buf: recv_data():148: buf 0x3fcbd564 double free
[00:00:48.719,000] <err> net_dns_dispatcher: DNS recv error (-2)
[00:00:48.722,000] <err> net_dns_dispatcher: DNS recv error (-2)
[00:00:48.815,000] <err> net_dns_dispatcher: DNS recv error (-2)
[00:00:48.822,000] <err> net_dns_dispatcher: DNS recv error (-2)
[00:00:49.019,000] <err> net_dns_dispatcher: DNS recv error (-2)
[00:00:49.042,000] <err> net_dns_dispatcher: DNS recv error (-2)
[00:00:49.046,000] <err> net_buf: recv_data():148: buf 0x3fcbd564 double free
[00:00:49.053,000] <err> net_dns_dispatcher: DNS recv error (-2)
[00:00:49.225,000] <err> net_dns_dispatcher: DNS recv error (-2)
[00:00:49.328,000] <err> net_dns_dispatcher: DNS recv error (-2)
```

jukkar, 2024-06-18:
Currently on vacation, I can investigate more next week.

rlubos, 2024-06-19:
@EricNRS I've identified the reason for double unref, and provided a fix: https://github.com/zephyrproject-rtos/zephyr/pull/74541
Please give it a try.

EricNRS, 2024-06-19:
Thanks @rlubos.  That fixed the double free.  I will run an overnight stress test to see if that takes care of the memory corruption issues that I was seeing.

Looking at the code, the double free does appear to be just a harmless nuisance warning since a lock was held that should have prevented the same packet from being reallocated.  I am just a bit perplexing as setting `CONFIG_MDNS_RESPONDER=n` and `CONFIG_DNS_RESOLVER=n` did pass overnight stress tests.  Let me know if you have any other ideas.

EricNRS, 2024-06-19:
Overnight stress test is still crashing due to memory corruption, but only once per hour instead of once per 15 minutes.  The corruption is random, so unsure of the root cause, but seeing a lot of DNS recv errors that I will look into and potentially open a different issue depending upon what I find.

```
[00:01:04.966,000] <err> net_dns_dispatcher: DNS recv error (-2)
[00:01:04.972,000] <err> net_dns_dispatcher: DNS recv error (-2)
[00:01:15.190,000] <err> net_dns_dispatcher: DNS recv error (-2)
[00:01:15.203,000] <err> net_dns_dispatcher: DNS recv error (-2)
[00:01:15.256,000] <err> net_dns_dispatcher: DNS recv error (-2)
[00:01:15.259,000] <err> net_dns_dispatcher: DNS recv error (-2)
[00:01:15.277,000] <err> net_dns_dispatcher: DNS recv error (-2)
[00:01:15.401,000] <err> net_dns_dispatcher: DNS recv error (-2)
[00:01:17.353,000] <err> net_dns_dispatcher: DNS recv error (-2)
[00:01:19.952,000] <err> net_dns_dispatcher: DNS recv error (-2)
[00:01:20.214,000] <err> net_dns_dispatcher: DNS recv error (-2)
```

In summary, the double free is fixed, so we can close this issue once the fix is merged and I will open a new issue if needed once I get to the bottom of the "DNS recv" failure.  Suggestions welcomed here or in Discord.

EricNRS, 2024-06-20:
Found the root cause.  The `serial_esp32_usb_irq_tx_enable()` enables the interrupts and then calls the interrupt handler on the current thread's context.  If an interrupt occurs before that completes, then it will pre-empt the thread, trash the ring buffer, and then return at which point the normal thread context throws an assertion.

I'll upload a potential fix soon.

PR #74541 (2024-06-21): net: lib: dns: Fix DNS dispatcher double net buf unref
The `dns_data` buffer, allocated by the DNS dispatcher was dereferenced twice - once in registered DNS handler, second time in the dispatcher itself.

Since the buffer was allocated by the dispatcher, and it's not really guaranteed that the buffer will be freed in the registered handler (this depends on the processing outcome, the function may return early w/o freeing the net buf in case of errors), it makes most sense for the dispatcher to keep ownership of the buffer. Hence, the registered handlers will no longer release the buffer provi [... cut]
closed 2024-06-21 by nashif