zephyrproject-rtos/zephyr #74397: Running Handsfree (BR/EDR) example on M5Stack Atom Echo.

xG3nesis, 2024-06-17:
Hello guys ! 
I'm attempting to run the `samples/bluetooth/handsfree` example on my M5Stack Atom Echo which has quite the same architecture than the M5Stack Atom Lite, I've just removed the useless lines on the devicetree to create one for my board. Everything is building and flashing okay, but when it runs on my board, I've the following error : 

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3ffbdb60,len:9216 
load:0x40080000,len:83440
1162 mmu set 00010000, pos 00010000
entry 0x4008e860
I (55) boot: ESP Simple boot
I (55) boot: compile time Jun 17 2024 10:45:22
W (55) boot: Unicore bootloader
I (55) spi_flash: detected chip: gd
I (57) spi_flash: flash io: dio
W (60) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (72) boot: chip revision: v1.1
I (75) boot.esp32: SPI Speed      : 40MHz
I (79) boot.esp32: SPI Mode       : DIO
I (82) boot.esp32: SPI Flash Size : 4MB
I (86) boot: Enabling RNG early entropy source...
[esp32] [INF] DRAM: lma 0x00001020 vma 0x3ffbdb60 len 0x2400   (9216)
[esp32] [INF] IRAM: lma 0x00003428 vma 0x40080000 len 0x145f0  (83440)
[esp32] [INF] padd: lma 0x00017a28 vma 0x00000000 len 0x85d0   (34256)
[esp32] [INF] IMAP: lma 0x00020000 vma 0x400d0000 len 0x2f868  (194664)
[esp32] [INF] padd: lma 0x0004f870 vma 0x00000000 len 0x788    (1928)
[esp32] [INF] DMAP: lma 0x00050000 vma 0x3f400000 len 0x3ef4   (16116)
[esp32] [INF] Image with 6 segments
[esp32] [INF] DROM segment: paddr=00050000h, vaddr=3f400000h, size=03EF4h ( 16116) map
[esp32] [INF] IROM segment: paddr=00020000h, vaddr=400d0000h, size=2F868h (194664) map

*** Booting Zephyr OS build v3.6.0-6150-gdf788d0b4130 ***
Bluetooth init failed (err 258)
```

This is what my DTS looks like : 
```/*
 * Copyright (c) 2024 Nikola Trifunovic <nikola@trifunovic.at>
 * Copyright (c) 2023 Benjamin Cabé <benjamin@zephyrproject.org>
 * Copyright (c) 2023 Martin Kiepfer <mrmarteng@teleschirm.org>
 *
 * SPDX-License-Identifier: Apache-2.0
 */
/dts-v1/;

#include <espressif/esp32/esp32_pico_d4.dtsi>
#include "m5stack_atom_echo-pinctrl.dtsi"
#include "grove_connectors.dtsi"
#include <zephyr/dt-bindings/input/input-event-codes.h>
#include <zephyr/dt-bindings/led/led.h>
#include <zephyr/dt-bindings/led/worldsemi_ws2812c.h>

/ {
	model = "M5Stack ATOM Echo PROCPU";
	compatible = "m5stack,m5stack-atom-echo";

	aliases {
		led-strip = &status_rgb_led;
		sw0 = &user_button_0;
		uart-0 = &uart0;
		i2c-0 = &i2c0;
		watchdog0 = &wdt0;
	};

	chosen {
		zephyr,sram = &sram0;
		zephyr,console = &uart0;
		zephyr,shell-uart = &uart0;
		zephyr,flash = &flash0;
		zephyr,code-partition = &slot0_partition;
		zephyr,bt-hci = &esp32_bt_hci;
	};

	gpio_keys {
		compatible = "gpio-keys";

		user_button_0: button_0 {
			label = "User button 0";
			gpios = <&gpio1 7 GPIO_ACTIVE_LOW>; // G39
			zephyr,code = <INPUT_KEY_0>;
		};
	};
};

&uart0 {
	status = "okay";
	current-speed = <115200>;
	pinctrl-0 = <&uart0_default>;
	pinctrl-names = "default";
};

&i2c0 {
	status = "okay";
	clock-frequency = <I2C_BITRATE_FAST>;
	pinctrl-0 = <&i2c0_default>;
	pinctrl-names = "default";
	sda-gpios = <&gpio0 25 GPIO_OPEN_DRAIN>;
	scl-gpios = <&gpio0 21 GPIO_OPEN_DRAIN>;

};

&i2c1 {
	status = "okay";
	clock-frequency = <I2C_BITRATE_FAST>;
	pinctrl-0 = <&i2c1_default>;
	pinctrl-names = "default";
	sda-gpios = <&gpio0 26 GPIO_OPEN_DRAIN>;
	scl-gpios = <&gpio1 0 GPIO_OPEN_DRAIN>;
};

&timer0 {
	status = "okay";
};

&timer1 {
	status = "okay";
};

&timer2 {
	status = "okay";
};

&timer3 {
	status = "okay";
};

&trng0 {
	status = "okay";
};

/* used for SK6812 */
&spi3 {
	#address-cells = <1>;
	#size-cells = <0>;
	status = "okay";
	line-idle-low;
	pinctrl-0 = <&spim3_ws2812_led>;
	pinctrl-names = "default";

	status_rgb_led: ws2812@0 {
		compatible = "worldsemi,ws2812-spi";
		reg = <0>;
		spi-max-frequency = <WS2812C_SPI_FREQ>;

		chain-length = <1>;
		color-mapping =	<LED_COLOR_ID_GREEN>,
			<LED_COLOR_ID_RED>,
			<LED_COLOR_ID_BLUE>;
		spi-one-frame = <WS2812C_ONE_FRAME>;
		spi-zero-frame = <WS2812C_ZERO_FRAME>;
	};
};

&gpio0 {
	status = "okay";
};

&gpio1 {
	status = "okay";
};

&wdt0 {
	status = "okay";
};

&flash0 {
	status = "okay";
	partitions {
		compatible = "fixed-partitions";
		#address-cells = <1>;
		#size-cells = <1>;

		/* Reserve 60kB for the bootloader */
		boot_partition: partition@1000 {
			label = "mcuboot";
			reg = <0x00001000 0x0000F000>;
			read-only;
		};

		/* Reserve 1024kB for the application in slot 0 */
		slot0_partition: partition@10000 {
			label = "image-0";
			reg = <0x00010000 0x00100000>;
		};

		/* Reserve 1024kB for the application in slot 1 */
		slot1_partition: partition@110000 {
			label = "image-1";
			reg = <0x00110000 0x00100000>;
		};

		/* Reserve 256kB for the scratch partition */
		scratch_partition: partition@210000 {
			label = "image-scratch";
			reg = <0x00210000 0x00040000>;
		};

		storage_partition: partition@250000 {
			label = "storage";
			reg = <0x00250000 0x00006000>;
		};
	};
};

&esp32_bt_hci {
	status = "okay";
};
```
I am running this example on the M5Stack Atom Echo, which features an ESP32-PICO-D4 and typically supports Bluetooth BR/EDR version 4.2.
[Atom Echo documentation](https://shop.m5stack.com/products/atom-echo-smart-speaker-dev-kit) !

**To Reproduce**
Steps to reproduce the behavior:
1.  `west build -p always -b m5stickc_plus/esp32/procpu samples/bluetooth/handsfree`
2. `west flash`
3. `west espressif monitor`

**Environment (please complete the following information):**
 - OS: Windows 10
 - Toolchain : Zephyr SDK
 - Version used : 0.16.8

I am unable to initialize the Bluetooth service on my ESP32 and I don't understand the source of the error. I'm quite a noob, so I'm really sorry if I've misunderstood a step or any kind of information that might help me to run the "handsfree" example.

Thank you in advance for your help ! 😄

jtyao, 2024-06-18:
Currently, ESP32 does not have good support for BR/EDR under Zephyr. You can modify the code as follows:
diff --git a/zephyr/esp32/include/bt/esp_bt.h b/zephyr/esp32/include/bt/esp_bt.h

index fb323c5532..04c9df0c7b 100644

--- a/zephyr/esp32/include/bt/esp_bt.h

+++ b/zephyr/esp32/include/bt/esp_bt.h

@@ -132,7 +132,7 @@ the adv packet will be discarded until the memory is restored. */

 #elif defined(CONFIG_BTDM_CTRL_MODE_BR_EDR_ONLY)

 #define BTDM_CONTROLLER_MODE_EFF                    ESP_BT_MODE_CLASSIC_BT

 #else

-#define BTDM_CONTROLLER_MODE_EFF                    ESP_BT_MODE_BLE

+#define BTDM_CONTROLLER_MODE_EFF                    ESP_BT_MODE_BTDM

 #endif

 

 #ifdef CONFIG_BTDM_CTRL_AUTO_LATENCY_EFF

@@ -181,7 +181,7 @@ the adv packet will be discarded until the memory is restored. */

 #define CONFIG_BTDM_CTRL_PCM_ROLE_EFF 0

 #define CONFIG_BTDM_CTRL_PCM_POLAR_EFF 0

 

-#ifdef CONFIG_BT_BREDR

+#ifdef CONFIG_BT_CLASSIC

 #define CONFIG_BTDM_CTRL_BR_EDR_MAX_SYNC_CONN_EFF CONFIG_BT_MAX_SCO_CONN

 #define CONFIG_BTDM_CTRL_BR_EDR_MAX_ACL_CONN_EFF 2

 #else

@@ -190,7 +190,6 @@ the adv packet will be discarded until the memory is restored. */

 #endif

 

 #define CONFIG_BTDM_CTRL_FULL_SCAN_SUPPORTED 1

-

 #define BT_CONTROLLER_INIT_CONFIG_DEFAULT() {                              \

     .controller_task_stack_size = ESP_TASK_BT_CONTROLLER_STACK,            \

     .controller_task_prio = ESP_TASK_BT_CONTROLLER_PRIO,                   \

xG3nesis, 2024-06-18:
Hello @jtyao ! 
Thank you very much, it's working fine right now ! 👍 
[image]

I have some additional questions regarding my main objective, which is to transfer audio recorded on my Atom Echo to my devices. Currently, the Zephyr stack doesn't support the I2S protocol on the ESP32, so I plan to use the ESP-IDF libraries with the Zephyr stack. My question is: can I use the Handsfree profile implemented on the Zephyr stack (BR/EDR) to transfer audio? Do you think that there would be any hardware/implementation issues ?

I understand that Zephyr has Audio LE libraries, but my board only supports Bluetooth 4.2, not the Bluetooth 5.2 version required for audio over Bluetooth Low Energy.

Thank you once more for your help, you are my sensei ! 🙏

xG3nesis, 2024-06-18:
Thank you for your prompt answer ! 
Is it supported on the ESP32-PICO-D4 board with bluetooth version 4.2 ? 🤔

jtyao, 2024-06-18:
It should be possible to run it, as the [OPUS-encoded](https://opus-codec.org/) audio data with a 16kHz sample rate and 20ms duration only amounts to 80 bytes, which is well within the bandwidth capabilities of BLE for transmission.