zephyrproject-rtos/zephyr #74417: CONFIG_WIFI_ESP32=y causes Kconfig warning on wemos ESP32-S2 Lolin Mini related to MBEDTLS

leriksen137, 2024-06-17:
# CONFIG_WIFI_ESP32=y causes Kconfig warning on wemos ESP32-S2 Lolin Mini related to MBEDTLS

## Bug
Can't build simple wifi application for the Wemos ESP32-S2 Lolin Mini board due to KConfig error. There seems to be an error where KConfig finds MBEDTLS defined in `zephyr/soc/nxp/` instead of `zephyr/soc/espressif`.

## Reproduction steps

Code for reproduction can be found here: https://github.com/leriksen137/config_wifi_esp32_does_not_build.git
This project has the same topology as the [zephyr example application](https://github.com/zephyrproject-rtos/example-application).

Follow [Zephyr getting started](https://docs.zephyrproject.org/latest/develop/getting_started/index.html).

```bash
git clone https://github.com/leriksen137/config_wifi_esp32_does_not_build.git
cd config_wifi_esp32_does_not_build
west init -l --mf west.yml manifest
west update
west blobs fetch hal_espressif
west build -b esp32s2_lolin_mini app
```
See [ESP32-S2 Lolin Mini](https://docs.zephyrproject.org/latest/boards/wemos/esp32s2_lolin_mini/doc/index.html) why west `blobs fetch hal_espressif` is run.

## Expected behavior
App should build without warnings.

## Actual behavior
Example console output:
```bash
west build -b esp32s2_lolin_mini app --pristine
-- west build: making build dir path\to\repo\config_wifi_esp32_does_not_build\build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: path/to/repo/config_wifi_esp32_does_not_build/app
-- CMake version: 3.27.9
-- Found Python3: C:/Python311/python.exe (found suitable version "3.11.9", minimum required is "3.8") found components: Interpreter
-- Cache files will be written to: path/to/repo/config_wifi_esp32_does_not_build/zephyr/.cache
-- Zephyr version: 3.7.0-rc1 (path/to/repo/config_wifi_esp32_does_not_build/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: esp32s2_lolin_mini, qualifiers: esp32s2
-- Found host-tools: zephyr 0.16.1 (~/zephyr-sdk-0.16.1)
-- Found toolchain: zephyr 0.16.1 (~/zephyr-sdk-0.16.1)
-- Found Dtc: C:/ProgramData/chocolatey/bin/dtc.exe (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: path/to/repo/config_wifi_esp32_does_not_build/zephyr/boards/wemos/esp32s2_lolin_mini/esp32s2_lolin_mini.dts
-- Found devicetree overlay: path/to/repo/config_wifi_esp32_does_not_build/app/boards/esp32s2_lolin_mini.overlay
-- Generated zephyr.dts: path/to/repo/config_wifi_esp32_does_not_build/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: path/to/repo/config_wifi_esp32_does_not_build/build/zephyr/include/generated/zephyr/devicetree_generated.h
-- Including generated dts.cmake file: path/to/repo/config_wifi_esp32_does_not_build/build/zephyr/dts.cmake

warning: MBEDTLS (defined at path/to/repo/config_wifi_esp32_does_not_build/zephyr/soc/nxp/imxrt\imxrt5xx\Kconfig.defconfig:24, path/to/repo/config_wifi_esp32_does_not_build/zephyr/soc/nxp/imxrt\imxrt6xx\Kconfig.defconfig:81, path/to/repo/config_wifi_esp32_does_not_build/zephyr/soc/nxp/imxrt/Kconfig.defconfig:99, modules\mbedtls\Kconfig:18) has direct dependencies (ENTROPY_GENERATOR && SOC_MIMXRT595S_CM33 && SOC_FAMILY_NXP_IMXRT) || (ENTROPY_GENERATOR && SOC_MIMXRT685S_CM33 && SOC_FAMILY_NXP_IMXRT) || (ENTROPY_GENERATOR && (SOC_SERIES_IMXRT10XX || SOC_SERIES_IMXRT11XX) && SOC_FAMILY_NXP_IMXRT) || 0 with value n, but is currently being y-selected by the following symbols:
 - WIFI_ESP32 (defined at drivers/wifi/esp32/Kconfig.esp32:3), with value y, direct dependencies DT_HAS_ESPRESSIF_ESP32_WIFI_ENABLED && !SMP && WIFI (value: y), and select condition DT_HAS_ESPRESSIF_ESP32_WIFI_ENABLED && !SMP && WIFI (value: y)Parsing path/to/repo/config_wifi_esp32_does_not_build/zephyr/Kconfig
Loaded configuration 'path/to/repo/config_wifi_esp32_does_not_build/zephyr/boards/wemos/esp32s2_lolin_mini/esp32s2_lolin_mini_defconfig'
Merged configuration 'path/to/repo/config_wifi_esp32_does_not_build/app/prj.conf'

error: Aborting due to Kconfig warnings

CMake Error at path/to/repo/config_wifi_esp32_does_not_build/zephyr/cmake/modules/kconfig.cmake:389 (message):
  command failed with return code: 1
Call Stack (most recent call first):
  path/to/repo/config_wifi_esp32_does_not_build/zephyr/cmake/modules/zephyr_default.cmake:132 (include)
  path/to/repo/smart-pump-iot-adapter/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  path/to/repo/smart-pump-iot-adapter/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:5 (find_package)

-- Configuring incomplete, errors occurred!
```
Note the reference to zephyr/soc/nxp in the warning message.

## Troubleshooting

I confirmed the output file in "path/to/repo/config_wifi_esp32_does_not_build/build/zephyr/zephyr.dts" contains:
```
	wifi: wifi {
		compatible = "espressif,esp32-wifi";
		status = "okay";
	};
```

## Severity
Severe. The project I'm working on can't be finished while this is unresolved.

sylvioalves, 2024-06-17:
@leriksen137 Can you try rebasing to latest main?

sylvioalves, 2024-06-17:
btw, I have just tested you branch in here. It builds properly, but requires prf.conf changes as suggested below:
```c
CONFIG_WIFI=y
CONFIG_NET_L2_WIFI_MGMT=y
CONFIG_NET_L2_ETHERNET=y
CONFIG_NETWORKING=y
CONFIG_NET_DHCPV4=y
CONFIG_NET_IPV4=y
CONFIG_NET_TCP=y
```

leriksen137, 2024-06-17:
> @leriksen137 Can you try rebasing to latest main?

I ran `west update` and it now points to commit 83958eb937d. Is that what you meant?

leriksen137, 2024-06-17:
> btw, I have just tested you branch in here. It builds properly, but requires prf.conf changes as suggested below:
> 
> ```c
> CONFIG_WIFI=y
> CONFIG_NET_L2_WIFI_MGMT=y
> CONFIG_NET_L2_ETHERNET=y
> CONFIG_NETWORKING=y
> CONFIG_NET_DHCPV4=y
> CONFIG_NET_IPV4=y
> CONFIG_NET_TCP=y
> ```

This does not contain [CONFIG_WIFI_ESP32](https://docs.zephyrproject.org/latest/kconfig.html#CONFIG_WIFI_ESP32). According to the documentation, it seems to me that I need to enable this. Not doing this caused some other issues in trying to get wifi running on the board I mentioned.

sylvioalves, 2024-06-17:
> This does not contain [CONFIG_WIFI_ESP32](https://docs.zephyrproject.org/latest/kconfig.html#CONFIG_WIFI_ESP32). According to the documentation, it seems to me that I need to enable this. Not doing [...]

You can add that, however, it is not needed. 
You have already enabled wifi in dts .overlay file. Then, Kconfig automatically adds that into build:
```c
menuconfig WIFI_ESP32
	bool "ESP32 SoC WiFi support"
	default y
	depends on DT_HAS_ESPRESSIF_ESP32_WIFI_ENABLED
```

Anyway, with the suggested `prj.conf` above, I was able to build the application.

sylvioalves, 2024-06-17:
```c
~/espressif/config_wifi_esp32_does_not_build (main) $ rm -rf build; west build -p -b esp32s2_lolin_mini app
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /home/sylvio/espressif/config_wifi_esp32_does_not_build/app
-- CMake version: 3.28.1
-- Found Python3: /usr/bin/python3 (found suitable version "3.10.12", minimum required is "3.8") found components: Interpreter 
-- Cache files will be written to: /home/sylvio/.cache/zephyr
-- Zephyr version: 3.6.99 (/home/sylvio/espressif/zephyr/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: esp32s2_lolin_mini, qualifiers: esp32s2
-- Found host-tools: zephyr 0.16.8 (/home/sylvio/zephyr-sdk-0.16.8)
-- Found toolchain: zephyr 0.16.8 (/home/sylvio/zephyr-sdk-0.16.8)
-- Found Dtc: /home/sylvio/zephyr-sdk-0.16.8/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /home/sylvio/espressif/zephyr/zephyr/boards/wemos/esp32s2_lolin_mini/esp32s2_lolin_mini.dts
-- Found devicetree overlay: /home/sylvio/espressif/config_wifi_esp32_does_not_build/app/boards/esp32s2_lolin_mini.overlay
-- Generated zephyr.dts: /home/sylvio/espressif/config_wifi_esp32_does_not_build/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/sylvio/espressif/config_wifi_esp32_does_not_build/build/zephyr/include/generated/zephyr/devicetree_generated.h
-- Including generated dts.cmake file: /home/sylvio/espressif/config_wifi_esp32_does_not_build/build/zephyr/dts.cmake
Parsing /home/sylvio/espressif/zephyr/zephyr/Kconfig
Loaded configuration '/home/sylvio/espressif/zephyr/zephyr/boards/wemos/esp32s2_lolin_mini/esp32s2_lolin_mini_defconfig'
Merged configuration '/home/sylvio/espressif/config_wifi_esp32_does_not_build/app/prj.conf'
Configuration saved to '/home/sylvio/espressif/config_wifi_esp32_does_not_build/build/zephyr/.config'
Kconfig header saved to '/home/sylvio/espressif/config_wifi_esp32_does_not_build/build/zephyr/include/generated/zephyr/autoconf.h'
-- Found GnuLd: /home/sylvio/zephyr-sdk-0.16.8/xtensa-espressif_esp32s2_zephyr-elf/xtensa-espressif_esp32s2_zephyr-elf/bin/ld.bfd (found version "2.38") 
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/sylvio/zephyr-sdk-0.16.8/xtensa-espressif_esp32s2_zephyr-elf/bin/xtensa-espressif_esp32s2_zephyr-elf-gcc
ESP-IDF path: /home/sylvio/espressif/zephyr/modules/hal/espressif
esptool path: /home/sylvio/espressif/zephyr/modules/hal/espressif/tools/esptool_py/esptool.py
-- Using ccache: /usr/bin/ccache
-- Configuring done (2.9s)
-- Generating done (0.1s)
-- Build files have been written to: /home/sylvio/espressif/config_wifi_esp32_does_not_build/build
-- west build: building application
[1/438] Preparing syscall dependency handling

[4/438] Generating include/generated/zephyr/version.h
-- Zephyr version: 3.6.99 (/home/sylvio/espressif/zephyr/zephyr), build: v3.6.0-6455-gf33d2c1251d8
[438/438] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:      535178 B    4194048 B     12.76%
     iram0_0_seg:       62528 B     224000 B     27.91%
     dram0_0_seg:      156184 B     224000 B     69.72%
     irom0_0_seg:      273034 B      7680 KB      3.47%
     drom0_0_seg:         64 KB         4 MB      1.56%
    rtc_iram_seg:          0 GB         8 KB      0.00%
    rtc_slow_seg:          0 GB         8 KB      0.00%
    rtc_data_seg:          0 GB         8 KB      0.00%
rtc_reserved_seg:          0 GB         0 GB
        IDT_LIST:          0 GB         8 KB      0.00%
Generating files from /home/sylvio/espressif/config_wifi_esp32_does_not_build/build/zephyr/zephyr.elf for board: esp32s2_lolin_mini
esptool.py v4.7.0
Creating esp32s2 image...
Image has only RAM segments visible. ROM segments are hidden and SHA256 digest is not appended.
Info: inserting 4 bytes padding between net_buf_pool_area and net_if_area
Merged 15 ELF sections
Successfully created esp32s2 image.
```

leriksen137, 2024-06-17:
> ```c
> ~/espressif/config_wifi_esp32_does_not_build (main) $ rm -rf build; west build -p -b esp32s2_lolin_mini app
> -- west build: generating a build system
> Loading Zephyr default modules (Zephyr [...]

Oh interesting. I found his warning while trying to make wifi work on this board. I guess my idea that it's related to CONFIG_WIFI_ESP32 not working was wrong then?

sylvioalves, 2024-06-17:
Not, it is not related I think. Have you tried building again?

PR #75680 (2024-07-11): soc: esp32s2: fix wifi clock gate
ESP32-S2 Wi-Fi clock is not initialized properly, causing instability when scanning or connecting to a SSID. In addition, SARADC was kept enabled to feed RND peripheral, adding more instability to Wi-Fi connection.

Fixes #74899
Fixes #74417
commit 43a55908 (2024-07-11): soc: esp32s2: fix wifi clock gate

ESP32-S2 Wi-Fi clock is not initialized properly, causing
instability when scanning or connecting to a SSID.

Fixes #74899
Fixes #74417
closed 2024-07-11 by nashif