zephyrproject-rtos/zephyr #74440: ESP32 ethernet interface do not start

sylvioalves, 2024-06-17:
**Describe the bug**
As discussed in #68907, ESP32 ethernet interface won't be up.

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. west build -p -b esp32_ethernet_kit/esp32/procpu samples/boards/esp32/ethernet/

**Expected behavior**
Interface up and running + IP addresses:
```c
*** Booting Zephyr OS build v3.7.0-rc1-19-g02ad7ef1f9cd ***
[00:00:01.114,000] <inf> net_config: Initializing network
[00:00:01.114,000] <inf> net_config: Waiting interface 1 (0x3ffb5f08) to be up...
[00:00:04.708,000] <inf> phy_mii: PHY (1) Link speed 100 Mb, full duplex

[00:00:04.708,000] <inf> net_config: Interface 1 (0x3ffb5f08) coming up
[00:00:04.708,000] <inf> net_config: Running dhcpv4 client...
[00:00:20.750,000] <inf> net_dhcpv4: Received: 192.168.68.124
[00:00:20.751,000] <inf> net_config: IPv4 address: 192.168.68.124
[00:00:20.751,000] <inf> net_config: Lease time: 7200 seconds
[00:00:20.751,000] <inf> net_config: Subnet: 255.255.255.0
[00:00:20.751,000] <inf> net_config: Router: 192.168.68.1
```

**Impact**
Showstopper as ethernet won´t  work.
```c
*** Booting Zephyr OS build v3.7.0-rc1-19-g02ad7ef1f9cd ***
[00:00:01.114,000] <inf> net_config: Initializing network
[00:00:01.114,000] <inf> net_config: Waiting interface 1 (0x3ffb5f08) to be up...
[00:00:04.708,000] <inf> phy_mii: PHY (1) Link speed 100 Mb, full duplex
...
...
```

ductuanhoang, 2024-06-18:
Hi @sylvioalves what revision "hal_espressif" do you use? what revision "hal_espressif" do you use? I have pulled this repo and I tried to build the firmware using RMII GPIO it got error, currently my revision in "west.yml" is:
```c
    - name: hal_espressif
      revision: 99dce9490ef4a84ddffe75c9d8853b30008c64bc
      path: modules/hal/espressif
      west-commands: west/west-commands.yml
      groups:
        - hal
```

sylvioalves, 2024-06-18:
@ductuanhoang check #74442 (WIP)

PR #74442 (2024-06-24): bugfix: driver: esp32: eth_mdio: fix clock and function calls
Both MDIO and Ethernet drivers share the same clock subsystem. After clock control update in #73807, clock_control_on() now returns -EALREADY for already initialized clock subsystem. As a result, ethernet driver won't initialize as needed. 
Due to the same clock control and hal updates, internal functions required updates.

Fixes #74440
Fixes #74632

As this PR updates hal_espressif, it also fixes #74632.
commit cc29fc11 (2024-06-24): driver: eth_esp32: fix shared clock check

Both MDIO and Ethernet drivers share the same clock subsystem.
After clock control update in #73807, clock_control_on()
now returns -EALREADY for already initialized clock subsystem.
As a result, ethernet driver won't initialize as needed.

Fixes #74440
closed 2024-06-24 by nashif