zephyrproject-rtos/zephyr #74485: GATT/CL/GAR/BV-10-C: IUT failed to send back to back ATT_READ_MULTIPLE_VARIABLE_LENGTH_REQ

chandra1787, 2024-06-18:
**Describe the bug**
According to GATT test spec zephyr(IUT) has to send two ATT_READ_MULTIPLE_VARIABLE_LENGTH_REQ commands to PTS, but in this case IUT sent only one ATT_READ_MULTIPLE_VARIABLE_LENGTH_REQ command to PTS. PTS waited for more than 3 mins and connection got disconnected. Same issue observed for GATT/CL/GAR/BV-11-C as well

**Platform: NRF52833-dk**

**To Reproduce**
Steps to reproduce the behavior:
1. Please execute the below 2 test cases with PTS GATT/CL/GAR/BV-10-C and GATT/CL/GAR/BV-11-C

**Expected behavior**
IUT has to send multiple ATT_READ_MULTIPLE_VARIABLE_LENGTH_REQ commands to PTS

**Impact**
Zephyr host stack conformance tests fails

**Logs and console output**
Please find the attached fail logs with [PTS]([url]([GATT_CL_GAR_BV_10_C_2024_06_14_15_59_12.zip](https://github.com/user-attachments/files/15886003/GATT_CL_GAR_BV_10_C_2024_06_14_15_59_12.zip)))

**Environment (please complete the following information):**
 - OS: Windows 10
 - Toolchain (e.g Zephyr SDK, ...)
 - Zephyr Main 
[GATT_CL_GAR_BV_10_C_2024_06_14_15_59_12.zip](https://github.com/user-attachments/files/15886014/GATT_CL_GAR_BV_10_C_2024_06_14_15_59_12.zip)
[GATT_CL_GAR_BV_11_C_2024_06_14_17_56_25.zip](https://github.com/user-attachments/files/15886015/GATT_CL_GAR_BV_11_C_2024_06_14_17_56_25.zip)

sjanc, 2024-06-21:
This requires implementing (rewriting) GATT client part in tester from legacy GATT service (ID 2), to GATT client service (ID 6).

legacy BTP GATT service is not able to handle multiple GATT operations due to design issues.

chandra1787, 2024-06-24:
Hi sjanc,
You mean to say Zephyr stack doesn't support these test cases and yet to implement?
Or if any code change required to pass these test cases, could you please share with us which file and where to modify.

sjanc, 2024-06-24:
I mean that tester application which is used by PTS automation (tests/bluetooth/tester) doesn't support required BTP interface to handle this.

It is not simple fix to re-implement gatt client to new BTP service, so for time being I'd suggest to run those tests manually (ie via shell) if you need them for qualification

chandra1787, 2024-06-24:
Could you please provide documents path how to run manually. We couldn't find any document or readme for manual execution.

jori-nordic, 2024-07-02:
It is an issue, but a feature request rather. Changing the labels.