zephyrproject-rtos/zephyr #74486: Stack conformance tests GAP/SEC/SEM/BV-24-C and GAP/SEC/SEM/BV-29-C fails

ranju28, 2024-06-18:
**Describe the bug**
Conformance test cases GAP/SEC/SEM/BV-24-C and GAP/SEC/SEM/BV-29-C are failing. 

IUT in Secure connections only mode is not rejecting GATT service request both before and after the authentication 
procedure when performed toward a peer not supporting LE Secure Connections. 

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using? Nordic's nrf52833-dk

**To Reproduce**
Steps to reproduce the behavior:
1. setup Zephyr environment and Auto PTS https://docs.zephyrproject.org/latest/connectivity/bluetooth/autopts/autopts-win10.html#setup-zephyr-project-for-windows
2. Build the auto pts tester app
3. Program the board 
4. start the auto PTS server pyton ./autoptsserver.py -S 65000 
5. Run the test cases python autoptsclient-zephyr.py zephyr-master ~/zephyrproject/build/zephyr/zephyr.elf -t COM46 -b cyw920829m2evk_02 -S 65000 -C 65001 --debug-logs --test-cases GAP/SEC/SEM/BV-24-C GAP/SEC/SEM/BV-29-C 
[GAP_SEC_SEM_BV_24_C_2024_06_14_13_51_49 (2).zip](https://github.com/user-attachments/files/15886031/GAP_SEC_SEM_BV_24_C_2024_06_14_13_51_49.2.zip)

**Expected behavior**
IUT should reject the gatt service request from Lower tester. 

**Impact**
Zephyr host stack conformance tests fails. 

**Logs and console output**
PTS logs are attached. 

**Environment (please complete the following information):**
 - OS: Windows 10
 - Toolchain (e.g Zephyr SDK, ...)
 - Zephyr Main

sjanc, 2024-06-18:
both tests are passing in our latest run

ranju28, 2024-06-18:
@sjanc may I know the PTS version you are using?

sjanc, 2024-06-18:
8.5.4 but those were also passing with 8.5.3

BTW I'm not sure about your setup but those tests are using sec_m1l4.conf overlay in our autopts bot setup

ranju28, 2024-06-20:
@sjanc  Thanks for the information.  After using the sec_m1l4.conf Overlay tests are passing. 
Closing the issue.

closed 2024-06-20 by jhedberg