zephyrproject-rtos/zephyr #74492: tfm: build fails (linux)

erwango, 2024-06-18:
**Describe the bug**
Can't build tfm on Ubuntu  (20.04)

**To Reproduce**
`west build -p auto -b mps2_an521_ns samples/tfm_integration/psa_protected_storage`

**Impact**
Blocker

**Logs and console output**
```
$ west build -p auto -b mps2_an521_ns samples/tfm_integration/psa_protected_storage
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /local/mcu/zephyrproject/zephyr/samples/tfm_integration/psa_protected_storage
-- CMake version: 3.21.4
-- Found Python3: /usr/bin/python3 (found suitable version "3.8.10", minimum required is "3.8") found components: Interpreter 
-- Cache files will be written to: /local/home/frq07517/.cache/zephyr
-- Zephyr version: 3.7.0-rc1 (/local/mcu/zephyrproject/zephyr)
-- Found west (found suitable version "1.0.0", minimum required is "0.14.0")
CMake Warning at /local/mcu/zephyrproject/zephyr/cmake/modules/boards.cmake:110 (message):
  Deprecated BOARD=mps2_an521_ns specified, board automatically changed to:
  mps2/an521/cpu0/ns.
Call Stack (most recent call first):
  /local/mcu/zephyrproject/zephyr/cmake/modules/zephyr_default.cmake:132 (include)
  /local/mcu/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /local/mcu/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:9 (find_package)

-- Board: mps2, qualifiers: an521/cpu0/ns
-- Found host-tools: zephyr 0.16.5 (/local/mcu/zephyrproject/sdk/zephyr-sdk-0.16.5)
-- Found toolchain: zephyr 0.16.5 (/local/mcu/zephyrproject/sdk/zephyr-sdk-0.16.5)
-- Found Dtc: /local/mcu/zephyrproject/sdk/zephyr-sdk-0.16.5/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /local/mcu/zephyrproject/zephyr/boards/arm/mps2/mps2_an521_cpu0_ns.dts
-- Generated zephyr.dts: /local/mcu/zephyrproject/zephyr/build/mps2_an521_ns/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /local/mcu/zephyrproject/zephyr/build/mps2_an521_ns/zephyr/include/generated/zephyr/devicetree_generated.h
-- Including generated dts.cmake file: /local/mcu/zephyrproject/zephyr/build/mps2_an521_ns/zephyr/dts.cmake
Parsing /local/mcu/zephyrproject/zephyr/Kconfig
Loaded configuration '/local/mcu/zephyrproject/zephyr/boards/arm/mps2/mps2_an521_cpu0_ns_defconfig'
Merged configuration '/local/mcu/zephyrproject/zephyr/samples/tfm_integration/psa_protected_storage/prj.conf'
Configuration saved to '/local/mcu/zephyrproject/zephyr/build/mps2_an521_ns/zephyr/.config'
Kconfig header saved to '/local/mcu/zephyrproject/zephyr/build/mps2_an521_ns/zephyr/include/generated/zephyr/autoconf.h'
-- Found GnuLd: /local/mcu/zephyrproject/sdk/zephyr-sdk-0.16.5/arm-zephyr-eabi/arm-zephyr-eabi/bin/ld.bfd (found version "2.38") 
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /local/mcu/zephyrproject/sdk/zephyr-sdk-0.16.5/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Using ccache: /usr/bin/ccache
CMake Warning at /local/mcu/zephyrproject/zephyr/modules/trusted-firmware-m/CMakeLists.txt:512 (message):
  TFM_DUMMY_PROVISIONING is enabled:

        The device will be provisioned using dummy keys and is NOT secure!
        This is not suitable for production

-- Configuring done
-- Generating done
-- Build files have been written to: /local/mcu/zephyrproject/zephyr/build/mps2_an521_ns
-- west build: building application
[1/231] Preparing syscall dependency handling

[2/231] Generating include/generated/zephyr/version.h
-- Zephyr version: 3.7.0-rc1 (/local/mcu/zephyrproject/zephyr), build: v3.7.0-rc1-8-gc52094facdde
[8/231] Generating ../../tfm/CMakeCache.txt
FAILED: tfm/CMakeCache.txt /local/mcu/zephyrproject/zephyr/build/mps2_an521_ns/tfm/CMakeCache.txt 
cd /local/mcu/zephyrproject/zephyr/build/mps2_an521_ns/tfm && /local/home/frq07517/.local/lib/python3.8/site-packages/cmake/data/bin/cmake -GNinja -DTFM_TOOLCHAIN_FILE=/local/mcu/zephyrproject/modules/tee/tf-m/trusted-firmware-m/toolchain_GNUARM.cmake -DCROSS_COMPILE=/local/mcu/zephyrproject/sdk/zephyr-sdk-0.16.5/arm-zephyr-eabi/bin/arm-zephyr-eabi -DCMAKE_BUILD_TYPE=MinSizeRel -DTFM_PLATFORM=arm/mps2/an521 -DCONFIG_TFM_BUILD_LOG_QUIET=ON -DCONFIG_TFM_MEMORY_USAGE_QUIET=OFF -DPython3_EXECUTABLE=/usr/bin/python3 -DCONFIG_TFM_SPM_BACKEND="IPC" -DBL2=TRUE -DMCUBOOT_IMAGE_VERSION_S=0.0.0+0 -DMCUBOOT_IMAGE_VERSION_NS=0.0.0+0 -DMCUBOOT_SIGNATURE_TYPE=EC-P256 -DMCUBOOT_USE_PSA_CRYPTO=ON -DMCUBOOT_BUILTIN_KEY=ON -DMCUBOOT_HW_KEY=OFF -DMCUBOOT_KEY_S=/local/mcu/zephyrproject/modules/tee/tf-m/trusted-firmware-m/bl2/ext/mcuboot/root-EC-P256.pem -DMCUBOOT_KEY_NS=/local/mcu/zephyrproject/modules/tee/tf-m/trusted-firmware-m/bl2/ext/mcuboot/root-EC-P256_1.pem -DTFM_ISOLATION_LEVEL=1 -DMCUBOOT_IMAGE_NUMBER=2 -DTFM_DUMMY_PROVISIONING=ON -DTFM_EXCEPTION_INFO_DUMP=ON -DMCUBOOT_LOG_LEVEL=INFO -DTFM_PARTITION_LOG_LEVEL=TFM_PARTITION_LOG_LEVEL_INFO -DTFM_SPM_LOG_LEVEL=TFM_SPM_LOG_LEVEL_INFO -DTFM_PARTITION_NS_AGENT_MAILBOX=OFF -DTFM_PARTITION_PROTECTED_STORAGE=ON -DTFM_PARTITION_INTERNAL_TRUSTED_STORAGE=ON -DTFM_PARTITION_CRYPTO=ON -DTFM_PARTITION_PLATFORM=ON -DTFM_PARTITION_INITIAL_ATTESTATION=OFF -DTFM_PARTITION_FIRMWARE_UPDATE=OFF -DMCUBOOT_PATH=/local/mcu/zephyrproject/bootloader/mcuboot -DCONFIG_TFM_ENABLE_FP=OFF -DTFM_TESTS_REVISION_CHECKS=OFF -DMBEDCRYPTO_PATH=/local/mcu/zephyrproject/modules/crypto/mbedtls /local/mcu/zephyrproject/modules/tee/tf-m/trusted-firmware-m
CMake Warning at cmake/version.cmake:32 (message):
  TFM_VERSION_MANUAL mismatches to actual TF-M version.  Please update
  TFM_VERSION_MANUAL in cmake/version.cmake
Call Stack (most recent call first):
  CMakeLists.txt:13 (include)

-- Found Git: /usr/bin/git (found version "2.25.1") 
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /local/mcu/zephyrproject/sdk/zephyr-sdk-0.16.5/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Populating cmsis
-- Configuring done
-- Generating done
-- Build files have been written to: /local/mcu/zephyrproject/zephyr/build/mps2_an521_ns/tfm/lib/ext/cmsis-subbuild
[1/9] Creating directories for 'cmsis-populate'
[1/9] Performing download step (git clone) for 'cmsis-populate'
Cloning into 'cmsis-src'...
remote: Enumerating objects: 28376, done.        
remote: Counting objects: 100% (3785/3785), done.        
remote: Compressing objects: 100% (951/951), done.        
remote: Total 28376 (delta 3214), reused 3146 (delta 2773), pack-reused 24591        
Receiving objects: 100% (28376/28376), 39.81 MiB | 18.20 MiB/s, done.
Resolving deltas: 100% (20864/20864), done.
HEAD is now at d0c460c1 CMSIS-Core(M): Add Cortex-M52 support : add core_m52.h
[2/9] Performing update step for 'cmsis-populate'
[4/9] No patch step for 'cmsis-populate'
[5/9] No configure step for 'cmsis-populate'
[6/9] No build step for 'cmsis-populate'
[7/9] No install step for 'cmsis-populate'
[8/9] No test step for 'cmsis-populate'
[9/9] Completed 'cmsis-populate'
'/usr/bin/git' 'apply' '--verbose' '/local/mcu/zephyrproject/modules/tee/tf-m/trusted-firmware-m/lib/ext/cmsis/0001-iar-Add-missing-v8.1m-check.patch'
Checking patch CMSIS/Core/Include/m-profile/cmsis_iccarm_m.h...
Applied patch CMSIS/Core/Include/m-profile/cmsis_iccarm_m.h cleanly.
-- Found Python3: /usr/bin/python3 (found version "3.8.10") found components: Interpreter 
terminate called after throwing an instance of 'std::out_of_range'
  what():  basic_string::erase: __pos (which is 12898) > this->size() (which is 129)
ninja: build stopped: subcommand failed.

```

**Environment (please complete the following information):**
 - OS: Linux (Ubuntu 22.04)
 - Toolchain: Zephyr SDK (0.16.5)
 - Commit SHA or Version used: v3.7.0-rc1-8-gc52094facdd

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

erwango, 2024-06-18:
Note: I've made the title very generic, don't hesitate to add meaningful information

ithinuel, 2024-06-18:
Thank you for reporting this issue.

It looks like the TF-M version your system is trying to build is not in sync with what the zephyr version you're trying to build expects.

Your `modules/tee/trusted-firmware-m` repository should be pointing at the commit https://github.com/zephyrproject-rtos/trusted-firmware-m/tree/bdc4df1734b870de43237d56eb1b2a7af016ee95

See https://github.com/zephyrproject-rtos/zephyr/blob/v3.7.0-rc1/west.yml#L329-L333

Can you confirm that your modules are up to date by running `west update` and checking the output for successful git operations?

ithinuel, 2024-06-18:
This is odd. TF-M should be in version 2.1 not 1.7.
That being said, it might be an issue with `git describe` picking an old tag as my `west init` (and subsequent `west update`) didn't pick the tags so my `git describe` only gives me the sha.

closed 2024-06-18 by erwango