zephyrproject-rtos/zephyr #74497: L2CAP/TIM/BV-03-C fails as IUT is not backing off on connection request collision

ranju28, 2024-06-18:
**Describe the bug**
when both IUT and PTS  send L2CAP_Connection_Req IUT shall reject with result=0x0004 in l2cap_conn_rsp packet. But IUT is accepting the connection. 
Running the test with PTS 8.5.3 and Zephyr Main.

Please also mention any information which could help others to understand
the problem you're facing:
What target platform are you using? -> Nordic's nrf52833-dk

**To Reproduce**
Steps to reproduce the behavior:
1. setup Zephyr environment and Auto PTS https://docs.zephyrproject.org/latest/connectivity/bluetooth/autopts/autopts-win10.html#setup-zephyr-project-for-windows
2. Build the auto pts tester app
3. Program the board
4. start the auto PTS server
5. Run the test case L2CAP/TIM/BV-03-C

**Expected behavior**
when both IUT and PTS  send L2CAP_Connection_Req IUT shall reject with result=0x0004 in l2cap_conn_rsp packet

**Impact**
[L2CAP_TIM_BV_03_C_2024_06_14_18_12_25.zip](https://github.com/user-attachments/files/15887982/L2CAP_TIM_BV_03_C_2024_06_14_18_12_25.zip)
Zephyr host stack conformance tests fails.

**Logs and console output**
PTS logs attached. 

**Environment (please complete the following information):**
 - Toolchain  Zephyr SDK 
 - Zephyr Main - OS: Windows 10

jori-nordic, 2024-06-19:
@sjanc Does that test also fail in our runs?

sjanc, 2024-06-19:
this test pass in weekly runs

ranju28, 2024-06-20:
with eatt_two_channels.conf overlay test is passing 
Closing the issue

closed 2024-06-20 by ranju28