zephyrproject-rtos/zephyr #74528: Bluetooth: host: Bluetooth initialization fails for all the BlueNRG boards

HoZHel, 2024-06-19:
**Describe the bug**
Bluetooth initialization fails and leads to fatal error for all the boards which are using any variant of ST BlueNRG BLE module.

 - target platform: `disco_l475_iot1`
 - This is a regression due to [#72090 ](https://github.com/zephyrproject-rtos/zephyr/pull/72090)

**To Reproduce**
Build the beacon sample available under `samples/bluetooth/beacon`.

**Expected behavior**
Successful Bluetooth initialization and then beacon advertising.

**Impact**
Fatal error for all the boards that are using BlueNRG BLE module.

**Logs and console output**
```
Booting Zephyr OS build zephyr-v3.1.0-33166-g97Starting Beacon Demo
ASSERTION FAIL [success] @ WEST_TOPDIR/zephyr/subsys/bluetooth/host/hci_core.c:397
        command opcode 0x1003 timeout
[00:00:00.005,000] <err> os: r0/a1:  0x00000003  r1/a2:  0x00000000  r2/a3:  0x00000003
[00:00:00.005,000] <err> os: r3/a4:  0x00000003 r12/ip:  0x00000005 r14/lr:  0x08002f19
[00:00:00.005,000] <err> os:  xpsr:  0x01000000
[00:00:00.005,000] <err> os: Faulting instruction address (r15/pc): 0x08002f26
[00:00:00.005,000] <err> os: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
[00:00:00.005,000] <err> os: Current thread: 0x20000c40 (unknown)
[00:00:00.062,000] <err> os: Halting system
```

**Environment:**
 - OS: Windows 10
 - Toolchain: Zephyr SDK
 - Zephyr main

jori-nordic, 2024-06-19:
I don't have HW to reproduce this. unassigned.

jori-nordic, 2024-06-19:
A hint is that after that patchset, the TX thread is gone and everything is now sent from the syswq. 
The error in the log seems like a timeout, and that can happen if the syswq is also used by the HCI driver.

msmttchr, 2024-06-20:
> A hint is that after that patchset, the TX thread is gone and everything is now sent from the syswq. The error in the log seems like a timeout, and that can happen if the syswq is also used by the HCI driver.

Apparently, the problem happens because the HCI driver initialization is performed in the work queue (`init_work`).
https://github.com/zephyrproject-rtos/zephyr/blob/14642091730b26abf49e9ca359fe993ac8f56137/subsys/bluetooth/host/hci_core.c#L86-L90
In the ST specific initialization, we have a call to `bt_hci_cmd_send`, this will add a new command to be executed in the `cmd_tx_queue`.
https://github.com/zephyrproject-rtos/zephyr/blob/7f0f0a4f97d6adb6c134db0788dd200fbfba8515/drivers/bluetooth/hci/hci_spi_st.c#L353

The remaining initialization will execute (still in the same work item) other HCI command using `bt_hci_cmd_send_sync`.
The first `bt_hci_cmd_send_sync ` will send the previous command in the queue and then try to send the current one, but this will fail since the previous command is not yet completed.

Looking deeper at function `process_pending_cmd`, we noticed that input parameter `timeout` is unused, is this intended?
https://github.com/zephyrproject-rtos/zephyr/blob/14642091730b26abf49e9ca359fe993ac8f56137/subsys/bluetooth/host/hci_core.c#L4627-L4637
We tentatively fixed the code, by using the timeout in `k_sem_take`, as follow:
`		if (k_sem_take(&bt_dev.ncmd_sem, timeout) == 0) {
`
@jori-nordic, is this an acceptable fix?

erwango, 2024-06-20:
Adding @jhedberg

jori-nordic, 2024-06-20:
@msmttchr Yes, it seems it's a bug from my part. Will you open a PR or should I?

Would it make sense for ST to send us some HW so we can run some sanity tests for reworks like those? I already got an eval kit from NXP for the same purpose.

msmttchr, 2024-06-20:
> @msmttchr Yes, it seems it's a bug from my part. Will you open a PR or should I?
> 
I think we can do on our side. I will ask @HoZHel to do it.
> Would it make sense for ST to send us some HW so we can run some sanity tests for reworks like those? I already got an eval kit from NXP for the same purpose.

Sure, definitively possible. I will try to grab a collection (2-3 at least) of ST supported boards and send them to you.
BTW: Thanks a lot for your quick reply.

closed 2024-06-24 by nashif