zephyrproject-rtos/zephyr #74533: zbus observer with a --sysbuild (for mcuboot) build breaks debug serial output on esp32s3_devkitm

JacoPlentify, 2024-06-19:
**Describe the bug**
First off, I believe this is specific to the esp32s3_devkitm and only occurs with a sysbuild (`west build -p -b esp32s3_devkitm --sysbuild`) build.
When I compile the below program with the zbus observer line uncommented, the serial console output in the application (not the bootloader) stops working (it does not print anything after the bootloader). I've confirmed (by stepping through the code) that the code does still execute; it is still hitting the `printk` and `k_sleep` lines repeatedly each second. So it is not like the application sits in some fault handler somewhere. There is just nothing coming over the serial. 
```
#include <stdio.h>
#include <zephyr/logging/log.h>
#include <zephyr/zbus/zbus.h>

struct my_zbus_struct_t {
	int my_int1;
	int my_int2;
};

ZBUS_CHAN_DEFINE(my_zbus_channel, struct my_zbus_struct_t, NULL, NULL, ZBUS_OBSERVERS_EMPTY,
		 ZBUS_MSG_INIT(.my_int1 = 1, .my_int2 = 2)
);

ZBUS_SUBSCRIBER_DEFINE(my_zbus_sub, 4);

/* This breaks the serial output for some reason, but only when building with --sysbuild */
// ZBUS_CHAN_ADD_OBS(my_zbus_channel, my_zbus_sub, 0);	

int main(void)
{
	while(1)
	{
		printk("Hello World! %s\n", CONFIG_BOARD);
		k_sleep(K_SECONDS(1));
	}
	return 0;
}
```

`west espressif monitor -p /dev/ttyUSB0` after flashing a build with the zbus observer line commented out:
```c
Executing task: west espressif monitor -p /dev/ttyUSB0 

--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd0000,len:0x1b04
load:0x403b6000,len:0x400
load:0x403b6400,len:0x48a0
load:0x403be000,len:0x1d00
entry 0x403b7e68
[esp32s3] [INF] MCUboot 2nd stage bootloader
[esp32s3] [INF] compiled at Jun 19 2024 07:46:26
[esp32s3] [INF] Chip revision: 0
[esp32s3] [INF] SPI Flash speed: 40MHz, mode: DIO, size: 8MB
[esp32s3] [INF] Image index: 0, Swap type: none
[esp32s3] [INF] Loading image 0 - slot 0 from flash, area id: 1
[esp32s3] [INF] Application start=403790C4h
[esp32s3] [INF] DRAM segment: paddr=00007E28h, vaddr=3FC8CB10h, size=00E5Ch (  3676) load
[esp32s3] [INF] IRAM segment: paddr=00002468h, vaddr=40374000h, size=059C0h ( 22976) load
[esp32s3] [INF] DROM segment: paddr=00010040h, vaddr=3C000040h, size=02430h (  9264) map
[esp32s3] [INF] IROM segment: paddr=00020000h, vaddr=42020000h, size=0556Bh ( 21867) map
*** Booting Zephyr OS build v3.6.0 ***
Hello World! esp32s3_devkitm
Hello World! esp32s3_devkitm
Hello World! esp32s3_devkitm
Hello World! esp32s3_devkitm
Hello World! esp32s3_devkitm
Hello World! esp32s3_devkitm
Hello World! esp32s3_devkitm
Hello World! esp32s3_devkitm
```

`west espressif monitor -p /dev/ttyUSB0` after flashing a build with the zbus observer line uncommented (note lack of `Hello World` and `Booting Zephyr` lines:
```text
Executing task: west espressif monitor -p /dev/ttyUSB0 

--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd0000,len:0x1b04
load:0x403b6000,len:0x400
load:0x403b6400,len:0x48a0
load:0x403be000,len:0x1d00
entry 0x403b7e68
[esp32s3] [INF] MCUboot 2nd stage bootloader
[esp32s3] [INF] compiled at Jun 19 2024 07:50:09
[esp32s3] [INF] Chip revision: 0
[esp32s3] [INF] SPI Flash speed: 40MHz, mode: DIO, size: 8MB
[esp32s3] [INF] Image index: 0, Swap type: none
[esp32s3] [INF] Loading image 0 - slot 0 from flash, area id: 1
[esp32s3] [INF] Application start=403790C4h
[esp32s3] [INF] DRAM segment: paddr=00007E30h, vaddr=3FC8CB10h, size=00E5Dh (  3677) load
[esp32s3] [INF] IRAM segment: paddr=00002470h, vaddr=40374000h, size=059C0h ( 22976) load
[esp32s3] [INF] DROM segment: paddr=00010040h, vaddr=3C000040h, size=02430h (  9264) map
[esp32s3] [INF] IROM segment: paddr=00020000h, vaddr=42020000h, size=0556Fh ( 21871) map
```

This is on zephyr v3.6.0
**To Reproduce**
Steps to reproduce the behavior:
1. Build above-mentioned application using sysbuild, with the zbus observer line uncommented: `west build -p -b esp32s3_devkitm --sysbuild`
2. flash application: `west flash`
3. cd to build directory of app: `cd build/app`
4. Start the monitor: `west espressif monitor -p /dev/ttyUSB0`
5. Observe serial output for application not being printed

**Expected behavior**
Expect zbus observer to not break the serial output.

**Impact**
Show stopper: Can not write integration tests that include any module that uses zbus. Will probably be able to redesign our system to not use zbus but I suspect a deeper underlining issue than zbus itself that will probably pop up somewhere else as well (see Additional context section below)

**Logs and console output**
See Describe the bug section above

**Environment (please complete the following information):**
 - Dev container docker file:
```dockerfile
FROM zephyrprojectrtos/ci:v0.26.9

RUN apt-get -y update && \
    apt-get install --no-install-recommends -y \
        python3 \
        python3-pip \
        python3-venv

RUN python3 -m pip install --upgrade pip

WORKDIR /workdir
RUN chown -R user:user /workdir

# Add user to dialout and set home permissions
RUN usermod -aG dialout user
RUN chown -R user:user /home/user

RUN chmod a+rwx /workdir

# Make 'user' default on launch
USER user

# Setup for esp-idf (mainly used for the debugging tools in esp-idf)
RUN mkdir -p $HOME/esp
RUN git clone -b v5.1.2 --recursive https://github.com/espressif/esp-idf.git $HOME/esp/esp-idf
RUN cd $HOME/esp/esp-idf && ./install.sh esp32s3

# Launch bash shell by default
CMD ["/bin/bash"]
```

- Zephyr V3.6.0
west.yml:
```yml
manifest:
  version: 0.8

  projects:
    - name: zephyr
      url: https://github.com/zephyrproject-rtos/zephyr
      revision: v3.6.0
      clone-depth: 10
      import:
        file: west.yml
        path-prefix: deps
        name-allowlist:
          - hal_espressif
          - tinycrypt
          - mcuboot
          - mcumgr
          - mbedtls

  self:
    path: app
```

**Additional context**
I think the issue has to do with how the linker allocates the iterable sections within the esp32 image. Both the zbus  and the zephyr logging system uses iterable sections to statically declare their respective zbus observers or log backends. When I place a breakpoint at zephyr/subsys/logging/log_core.c:304 I observe the following:

With zbus observer line uncommented:
[image]

while with the zbus observer line commented out I get:
[image]

Note that without a zbus observer, we have a properly defined `log_backend` but when we add the zbus observer all the parameters for our `log_backend` are defined as 0x0.

sylvioalves, 2024-06-19:
@JacoPlentify Does it work without using mcuboot (--sysbuild) and keep zbus as is?

JacoPlentify, 2024-06-19:
@sylvioalves Yes it does, which is quite weird.

sylvioalves, 2024-06-19:
We have to address this issue.  This is section alignment stuff.
```c
#undef SECTION_PROLOGUE
#define SECTION_PROLOGUE SECTION_DATA_PROLOGUE
```

Can you add the above in this line: 
https://github.com/zephyrproject-rtos/zephyr/blob/main/soc/espressif/esp32s3/default.ld#L43

JacoPlentify, 2024-06-19:
@sylvioalves Switching to main, the problem persists but adding the changes you've suggested to main fixes the problem :)

brandon-exact, 2024-06-20:
I also discovered the same issue. Hoping this fix gets into the upcoming release :)

EricNRS, 2024-06-27:
I ran into the same issue with the `net_socket_register` iterable section for ZTLS due to the alignment being off by 4 bytes.

The `SECTION_DATA_PROLOGUE` prolog fix from @sylvioalves fixed the offset issue.
```
#undef SECTION_PROLOGUE
#define SECTION_PROLOGUE SECTION_DATA_PROLOGUE
```

Here is the debugging I went through in case that adds some value.  ESP32-S3 system is crashing on boot-up due to the ZTLS not-socket-register handle is set to 0:

```
(gdb) print *(struct net_socket_register*)&__net_socket_register_45_tls
$9 = {family = 0, is_offloaded = 236, is_supported = 0x420484c4 <ztls_socket>, handler = 0x0}
```

The address was actually shifted by 4 bytes:
```
Offsetting the address by 4 bytes results in the correct value:
(gdb) print *(struct net_socket_register*)(0x3c0f5c00-4)
$8 = {family = 0, is_offloaded = false, is_supported = 0x420459ec <tls_is_supported>, handler = 0x420484c4 <ztls_socket>}
```

The ELF file confirms that the data in memory was shifted:
```
${ZEPHYR_SDK_INSTALL_DIR}/xtensa-espressif_esp32s3_zephyr-elf/bin/xtensa-espressif_esp32s3_zephyr-elf-objcopy -O binary --only-section=net_socket_register_area build/zephyr/zephyr.elf net_socket_register_area.bin

hexdump -v -e '"%08x "' net_socket_register_area.bin
0000000 00000000 420459ec 420484c4
00000000 00000000 4203ffd0 42040ef8
```

I added an `ASSERT()` to the code before the handle was called as part of the debugging and the code started working correctly again.  Note that adding the assert shifted the net_socket_register segment by 0x20 bytes:
```
(gdb) print *sock_family
$2 = {family = 0, is_offloaded = false, is_supported = 0x42045a10 <tls_is_supported>, handler = 0x420484e8 <ztls_socket>}

# Note that the address is 0x20 bytes later, so there is a shift in the data
gdb) x/24x &__net_socket_register_45_tls
0x3c0f5c20 <__net_socket_register_45_tls>:      0x00000000      0x00000000      0x42045a10      0x420484e8
0x3c0f5c30 <__net_socket_register_50_af_inet46>:        0x00000000      0x00000000      0x4203ffd8      0x42040f00
0x3c0f5c40 <resolve_svc>:       0x00000000      0x3fc9fa94      0x00000001      0x3fcc4eec
```

I then added an `align(16)` for the initial work-around, but I think the `SECTION_DATA_PROLOGUE` fix is the proper one.
```
diff --git a/soc/espressif/esp32s3/default.ld b/soc/espressif/esp32s3/default.ld
index f93569bd167..dc59ffeddaf 100644
--- a/soc/espressif/esp32s3/default.ld
+++ b/soc/espressif/esp32s3/default.ld
@@ -849,7 +849,7 @@ SECTIONS
     . = ALIGN(4);
     *(.rodata_wlog)
     *(.rodata_wlog*)
-    . = ALIGN(4);
+    . = ALIGN(16);
   } GROUP_DATA_LINK_IN(RODATA_REGION, ROMABLE_REGION)
 
   #include <zephyr/linker/cplusplus-rom.ld>
```

PR #75101 (2024-06-29): soc: esp32xx: always use section prologue with alignment
ESP32 requires proper alignment between sections. There are some scenarios, as reported in #74533, that the section can get shifted, causing runtime failure.
Making sure SECTION_PROLOGUE is used with ALIGN_WITH_INPUT will guarantee its consistency.

Fixes #74533
commit f65770c1 (2024-06-29): soc: esp32xx: always use section prologue with alignment

ESP32 requires proper alignment between sections. There are some
scenarios, as reported in #74533, that the section can
get shifted, causing runtime failure.
Making sure SECTION_PROLOGUE is used with ALIGN_WITH_INPUT
will guarantee its consistency.

Fixes #74533
closed 2024-06-29 by nashif