zephyrproject-rtos/zephyr #74536: samples: subsys: llext: modules: sample.llext.modules.module_build reports overflow

katgiadla, 2024-06-19:
**Describe the bug**
Sample `samples/subsys/llext/modules/sample.llext.modules.module_build` report overflow on RAM region.

Observed for:
- `nrf52dk/nrf52832`
- `nrf52840dk/nrf52840`
- `nrf5340dk/nrf5340/cpuapp`
- `nrf5340dk/nrf5340/cpuapp/ns`
- `nrf5340dk/nrf5340/cpunet`
- `nrf54l15pdk/nrf54l15/cpuapp`
- `nrf9160dk/nrf9160`
- `nrf9160dk/nrf9160/ns`

**To Reproduce**
Steps to reproduce the behavior:
1. have `nrf52840dk/nrf52840` connected
2. go to your zephyr dir
3. call `./scripts/twister -T samples/subsys/llext/modules -p nrf52840dk/nrf52840 --device-testing --device-serial /dev/ttyACM0 -v --inline-logs --west-flash="--erase"`
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
*** Booting Zephyr OS build v3.7.0-rc1-80-g694428947ae3 ***
[00:00:00.254,760] <inf> app: Calling hello world as a module
[00:00:00.260,986] <err> os: ***** USAGE FAULT *****
[00:00:00.266,571] <err> os:   Stack overflow (context area not valid)
[00:00:00.273,742] <err> os: r0/a1:  0xb00bd09e  r1/a2:  0xacfbbeff  r2/a3:  0x00002256
[00:00:00.282,348] <err> os: r3/a4:  0x29000200 r12/ip:  0x200112c0 r14/lr:  0x00009068
[00:00:00.290,985] <err> os:  xpsr:  0x00000000
[00:00:00.296,173] <err> os: Faulting instruction address (r15/pc): 0x000003bb
[00:00:00.304,016] <err> os: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
[00:00:00.311,889] <err> os: Current thread: 0x20000658 (unknown)
[00:00:00.318,603] <err> os: Halting system
```

Sample do not build on `nrf52dk/nrf52832` and `nrf5340dk/nrf5340/cpunet`:
```
opt/zephyr-sdk/arm-zephyr-eabi/bin/lib/gcc/arm-zephyr-eabi/12.2.0/arm-zephyr-eabi/bin/ld.bfd: zephyr/zephyr_pre0.elf section `noinit' will not fit in region `RAM'
opt/zephyr-sdk/arm-zephyr-eabi/bin/lib/gcc/arm-zephyr-eabi/12.2.0/arm-zephyr-eabi/bin/ld.bfd: region `RAM' overflowed by 8752 bytes
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain: Zephyr SDK 0.16.5-1
 - Commit SHA or Version used: `v3.7.0-rc1-24-g0561d74d31c4`

**Additional context**
Probably introduced by PR #74060

hakehuang, 2024-06-20:
same prblem with many NXP platforms

<html>
<body>
<span><span class="ui-provider a b c d e f g h i j k l m n o p q r s t u v w x y z ab ac ae af ag ah ai aj ak" dir="ltr">
mimxrt1020_evk
--
mimxrt1040_evk
twr_kv58f220m
mimxrt1050_evk
mimxrt1160_evk/mimxrt1166/cm7
lpcxpresso55s06
frdm_mcxn947/mcxn947/cpu0
mimxrt685_evk/mimxrt685s/cm33
frdm_rw612
frdm_k64f

</span></span>
</body>
</html>

``
*** Booting Zephyr OS build v3.7.0-rc1-2-g4d0a8c1e8b83 ***
[01:05:11.019,689] <inf> app: Calling hello world as a module
[01:05:11.026,024] <err> os: ***** USAGE FAULT *****
[01:05:11.031,647] <err> os:   Stack overflow (context area not valid)
[01:05:11.038,973] <err> os: r0/a1:  0x00000000  r1/a2:  0x00000000  r2/a3:  0x18003796
[01:05:11.047,751] <err> os: r3/a4:  0x21000200 r12/ip:  0x00000000 r14/lr:  0x1800ac58
[01:05:11.056,547] <err> os:  xpsr:  0x00000000
[01:05:11.061,830] <err> os: Faulting instruction address (r15/pc): 0x00000001
[01:05:11.069,843] <err> os: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
[01:05:11.077,848] <err> os: Current thread: 0x30040ef8 (unknown)
[01:05:11.084,709] <err> os: Halting system
```

mathieuchopstm, 2024-06-20:
It does look like a stack overflow during the log/printing calls.
Using `west debug` and a well-placed breakpoint, the following can be observed:
```
(gdb) bt
#0  out_func (c=..., ctx=...) at zephyrproject/zephyr/subsys/logging/log_output.c:112
#1  0x08006594 in cbputc (c=..., _s=...) at zephyrproject/zephyr/lib/libc/picolibc/libc-hooks.c:88
#2  0x08005d7c in __l_vfprintf () at zephyrproject/zephyr/kernel/mempool.c:40
#3  0x080020be in cbvprintf (out=..., ctx=..., fp=..., ap=...) at zephyrproject/zephyr/lib/libc/picolibc/libc-hooks.c:99
#4  0x0800112c in print_formatted (output=..., fmt=...) at zephyrproject/zephyr/subsys/logging/log_output.c:144
#5  0x0800126c in timestamp_print (timestamp=..., flags=..., output=...) at zephyrproject/zephyr/subsys/logging/log_output.c:290
#6  prefix_print (tid=..., func_on=..., level=..., source=..., domain=..., timestamp=..., flags=..., output=...) at zephyrproject/zephyr/subsys/logging/log_output.c:643
#7  log_output_process (output=..., timestamp=..., domain=..., source=..., tid=..., level=..., package=..., data=..., data_len=..., flags=...) at zephyrproject/zephyr/subsys/logging/log_output.c:684
#8  0x080064a4 in log_output_msg_process (output=..., msg=..., flags=...) at zephyrproject/zephyr/subsys/logging/log_output.c:727
#9  0x08000ed6 in log_backend_msg_process (msg=..., backend=...) at zephyrproject/zephyr/include/zephyr/logging/log_backend.h:180
#10 msg_process (msg=...) at zephyrproject/zephyr/subsys/logging/log_core.c:502
#11 msg_commit (msg=..., buffer=...) at zephyrproject/zephyr/subsys/logging/log_core.c:662
#12 z_log_msg_commit (msg=...) at zephyrproject/zephyr/subsys/logging/log_core.c:676
#13 0x08000fe8 in z_log_msg_runtime_vcreate (domain_id=..., source=..., level=..., data=..., dlen=..., package_flags=..., fmt=..., ap=...) at zephyrproject/zephyr/subsys/logging/log_msg.c:389
#14 0x08005dce in z_log_msg_runtime_create (domain_id=..., source=..., level=..., data=..., dlen=..., package_flags=..., fmt=..., package_flags=..., dlen=..., data=..., domain_id=...) at zephyrproject/zephyr/include/zephyr/logging/log_msg.h:714
#15 0x080004b6 in main () at zephyrproject/zephyr/samples/subsys/llext/modules/src/main_module.c:20
(gdb) p $sp
$1 = (void *) 0x20011288 <z_main_stack+64>
```
i.e., as early as `main`, only 64 bytes free stack bytes remain!

> Can you create a fix PR so that I can test on all my failure platforms?
Sure, doing it right now.

PR #74582 (2024-06-21): samples: llext: modules: increase stack size to prevent stack overflow
Make main stack bigger for the `samples/subsys/llext/modules` sample to prevent stack overflow. On STM32 Nucleo-F401RE, the test only has 64 bytes of free stack left as early as `main` and the sample crashes with a USAGE FAULT due to the stack overflow.

Fixes #74536.
closed 2024-06-21 by nashif