zephyrproject-rtos/zephyr #74545: Missing openocd.cfg in nucleo_h563zi

QuentinCaldeira-eaton, 2024-06-19:
**Is your enhancement proposal related to a problem? Please describe.**
Almost all stm32 boards have an openocd.cfg file in the support folder, but nucleo_h563zi does not.

**Describe the solution you'd like**
Add config files in boards 

**Describe alternatives you've considered**
None

FRASTM, 2024-06-19:
One reason is that openocd is not mentionned as a runner of the stm32h563 nucleo neither stm32h573i_dk
The current release of the  openocd [openocd/tcl/target](https://github.com/zephyrproject-rtos/openocd/tree/zephyr-20220611)/[tcl](https://github.com/zephyrproject-rtos/openocd/tree/zephyr-20220611/tcl)  does not include the stm32h5x.cfg

erwango, 2024-06-27:
@FRASTM Would you min adding a PR to fix that directly in boards ?
We can add the cfg file and document how to use st openocd the same way we did for wba: https://github.com/zephyrproject-rtos/zephyr/commit/94477c099c338278767fa8285562c51d82717894

@QuentinCaldeira-eaton Pleas e note that, however, pyocd is supported and provide the same service as openocd.

FRASTM, 2024-07-02:
> @FRASTM Would you min adding a PR to fix that directly in boards ? We can add the cfg file and document how to use st openocd the same way we did for wba: [94477c0](https://github.com/zephyrproject-rtos/zephyr/commit/94477c099c338278767fa8285562c51d82717894)
> 
> @QuentinCaldeira-eaton Pleas e note that, however, pyocd is supported and provide the same service as openocd.

--> see PR https://github.com/zephyrproject-rtos/zephyr/pull/75308

closed 2024-07-30 by fabiobaltieri