zephyrproject-rtos/zephyr #74575: tests: net: conn_mgr_monitor: net.conn_mgr.nodad and net.conn_mgr.dad fails on NXP platforms

hakehuang, 2024-06-20:
**Describe the bug**

tests/net/conn_mgr_monitor/net.conn_mgr.nodad and tests/net/conn_mgr_monitor/net.conn_mgr.dad fail on all NXP platforms

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 frdm_k64f, mimxrt1050_evk
 - What have you tried to diagnose or workaround this issue?
 git bisect
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
after merge this chnage:

976e0a49e44ac161e9e7645185d7075c3d86d52e

**To Reproduce**
Steps to reproduce the behavior:
1. scripts/twister -p frdm_k64f -T tests/net/conn_mgr_monitor 

**Expected behavior**
test PASS

**Impact**
net function

**Logs and console output**

```
*** Booting Zephyr OS build v3.7.0-rc1-2-g4d0a8c1e8b83 ***
Running TESTSUITE conn_mgr_monitor
===================================================================
START - test_DAD
SKIP - test_DAD in 0.002 seconds
===================================================================
START - test_cycle_ready_CC
NET_EVENT_L4_CONNECTED
NET_EVENT_L4_IPV4_CONNECTED
NET_EVENT_L4_DISCONNECTED
Assertion failed at WEST_TOPDIR/zephyr/tests/net/conn_mgr_monitor/src/main.c:289: cycle_ready_ifaces: (stats.event_count_gen not equal to 0)
No events should be fired if connectivity availability did not change.
NET_EVENT_L4_IPV4_DISCONNECTED
NET_EVENT_L4_CONNECTED
NET_EVENT_L4_IPV4_CONNECTED
FAIL - test_cycle_ready_CC in 0.037 seconds
===================================================================
START - test_cycle_ready_CIC
NET_EVENT_L4_DISCONNECTED
cycle_ignored_iface
NET_EVENT_L4_IPV4_DISCONNECTED
Expect no events.
Assertion failed at WEST_TOPDIR/zephyr/tests/net/conn_mgr_monitor/src/main.c:343: cycle_ignored_iface: (stats.event_count_ipv4 not equal to 0)
No events should be fired if connectivity availability did not change.
FAIL - test_cycle_ready_CIC in 0.030 seconds
===================================================================
START - test_cycle_ready_CINC
cycle_ignored_iface
Expect no events.
NET_EVENT_L4_CONNECTED
Assertion failed at WEST_TOPDIR/zephyr/tests/net/conn_mgr_monitor/src/main.c:341: cycle_ignored_iface: (stats.event_count_gen not equal to 0)
No events should be fired if connectivity availability did not change.
FAIL - test_cycle_ready_CINC in 0.096 seconds
===================================================================
START - test_cycle_ready_CNC
NET_EVENT_L4_CONNECTED
Assertion failed at WEST_TOPDIR/zephyr/tests/net/conn_mgr_monitor/src/main.c:272: cycle_ready_ifaces: (stats.conn_iface_gen not equal to ifa)
ifa should be blamed.
NET_EVENT_L4_CONNECTED
NET_EVENT_L4_IPV4_CONNECTED
FAIL - test_cycle_ready_CNC in 0.064 seconds
[... 256 lines cut ...]
- FAIL - [conn_mgr_monitor.test_cycle_ready_CC] duration = 0.031 seconds
- FAIL - [conn_mgr_monitor.test_cycle_ready_CIC] duration = 0.140 seconds
- FAIL - [conn_mgr_monitor.test_cycle_ready_CINC] duration = 0.207 seconds
- PASS - [conn_mgr_monitor.test_cycle_ready_CNC] duration = 0.069 seconds
- PASS - [conn_mgr_monitor.test_cycle_ready_NCC] duration = 0.016 seconds
- FAIL - [conn_mgr_monitor.test_cycle_ready_NCIC] duration = 0.369 seconds
- FAIL - [conn_mgr_monitor.test_cycle_ready_NCINC] duration = 0.369 seconds
- PASS - [conn_mgr_monitor.test_cycle_ready_NCNC] duration = 0.016 seconds
- FAIL - [conn_mgr_monitor.test_cycle_states_connected_ipv46] duration = 0.028 seconds
- FAIL - [conn_mgr_monitor.test_cycle_states_connected_ipv64] duration = 0.150 seconds
- FAIL - [conn_mgr_monitor.test_cycle_states_simple_ipv46] duration = 0.029 seconds
- FAIL - [conn_mgr_monitor.test_cycle_states_simple_ipv64] duration = 0.149 seconds
- FAIL - [conn_mgr_monitor.test_ignore_while_ready] duration = 0.142 seconds
- PASS - [conn_mgr_monitor.test_ignores] duration = 0.010 seconds
------ TESTSUITE SUMMARY END ------
===================================================================
RunID: 365d7add00f824592955c7bd44164bb6
PROJECT EXECUTION FAILED

```

**Environment (please complete the following information):**
 - OS: (e.g. Linux )
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used:  v3.7.0-rc1-2-g4d0a8c1e8b83

hakehuang, 2024-06-20:
last pass log
```
2024-06-02 00:15:34,055 - twister - DEBUG - DEVICE: *** Booting Zephyr OS build v3.6.0-5112-g4a6b8f27b8a2 ***
2024-06-02 00:15:34,058 - twister - DEBUG - DEVICE: Running TESTSUITE conn_mgr_monitor
2024-06-02 00:15:34,064 - twister - DEBUG - DEVICE: ===================================================================
2024-06-02 00:15:34,065 - twister - DEBUG - DEVICE: START - test_DAD
2024-06-02 00:15:34,181 - twister - DEBUG - DEVICE: PASS - test_DAD in 0.113 seconds
2024-06-02 00:15:34,185 - twister - DEBUG - -- west flash: using runner pyocd
-- runners.pyocd: Flashing file: /home/ubuntu/nxp/frdm_k64f/temp/frdm_k64f_twister_test_only/twister-out/frdm_k64f/tests/net/conn_mgr_monitor/net.conn_mgr.dad/zephyr/zephyr.bin
[---|---|---|---|---|---|---|---|---|----]
[========================================]

2024-06-02 00:15:34,187 - twister - DEBUG - DEVICE: ===================================================================
2024-06-02 00:15:34,190 - twister - DEBUG - DEVICE: START - test_cycle_ready_CC
2024-06-02 00:15:34,201 - twister - DEBUG - DEVICE: PASS - test_cycle_ready_CC in 0.008 seconds
2024-06-02 00:15:34,207 - twister - DEBUG - DEVICE: ===================================================================
2024-06-02 00:15:34,210 - twister - DEBUG - DEVICE: START - test_cycle_ready_CIC
2024-06-02 00:15:34,230 - twister - DEBUG - DEVICE: PASS - test_cycle_ready_CIC in 0.018 seconds
2024-06-02 00:15:34,237 - twister - DEBUG - DEVICE: ===================================================================
2024-06-02 00:15:34,239 - twister - DEBUG - DEVICE: START - test_cycle_ready_CINC
2024-06-02 00:15:34,260 - twister - DEBUG - DEVICE: PASS - test_cycle_ready_CINC in 0.018 seconds
2024-06-02 00:15:34,266 - twister - DEBUG - DEVICE: ===================================================================
2024-06-02 00:15:34,269 - twister - DEBUG - DEVICE: START - test_cycle_ready_CNC
2024-06-02 00:15:34,281 - twister - DEBUG - DEVICE: PASS - test_cycle_ready_CNC in 0.008 seconds
2024-06-02 00:15:34,287 - twister - DEBUG - DEVICE: ===================================================================
2024-06-02 00:15:34,289 - twister - DEBUG - DEVICE: START - test_cycle_ready_NCC
2024-06-02 00:15:34,301 - twister - DEBUG - DEVICE: PASS - test_cycle_ready_NCC in 0.008 seconds
2024-06-02 00:15:34,308 - twister - DEBUG - DEVICE: ===================================================================
2024-06-02 00:15:34,310 - twister - DEBUG - DEVICE: START - test_cycle_ready_NCIC
2024-06-02 00:15:34,332 - twister - DEBUG - DEVICE: PASS - test_cycle_ready_NCIC in 0.018 seconds
2024-06-02 00:15:34,337 - twister - DEBUG - DEVICE: ===================================================================
2024-06-02 00:15:34,340 - twister - DEBUG - DEVICE: START - test_cycle_ready_NCINC
2024-06-02 00:15:34,361 - twister - DEBUG - DEVICE: PASS - test_cycle_ready_NCINC in 0.018 seconds
2024-06-02 00:15:34,367 - twister - DEBUG - DEVICE: ===================================================================
2024-06-02 00:15:34,370 - twister - DEBUG - DEVICE: START - test_cycle_ready_NCNC
2024-06-02 00:15:34,382 - twister - DEBUG - DEVICE: PASS - test_cycle_ready_NCNC in 0.008 seconds
2024-06-02 00:15:34,388 - twister - DEBUG - DEVICE: ===================================================================
2024-06-02 00:15:34,392 - twister - DEBUG - DEVICE: START - test_cycle_states_connected_ipv46
2024-06-02 00:15:34,631 - twister - DEBUG - DEVICE: PASS - test_cycle_states_connected_ipv46 in 0.235 seconds
2024-06-02 00:15:34,637 - twister - DEBUG - DEVICE: ===================================================================
2024-06-02 00:15:34,640 - twister - DEBUG - DEVICE: START - test_cycle_states_connected_ipv64
2024-06-02 00:15:34,880 - twister - DEBUG - DEVICE: PASS - test_cycle_states_connected_ipv64 in 0.234 seconds
2024-06-02 00:15:34,885 - twister - DEBUG - DEVICE: ===================================================================
2024-06-02 00:15:34,889 - twister - DEBUG - DEVICE: START - test_cycle_states_simple_ipv46
2024-06-02 00:15:35,128 - twister - DEBUG - DEVICE: PASS - test_cycle_states_simple_ipv46 in 0.235 seconds
2024-06-02 00:15:35,134 - twister - DEBUG - DEVICE: ===================================================================
2024-06-02 00:15:35,137 - twister - DEBUG - DEVICE: START - test_cycle_states_simple_ipv64
2024-06-02 00:15:35,376 - twister - DEBUG - DEVICE: PASS - test_cycle_states_simple_ipv64 in 0.234 seconds
2024-06-02 00:15:35,382 - twister - DEBUG - DEVICE: ===================================================================
2024-06-02 00:15:35,385 - twister - DEBUG - DEVICE: START - test_ignore_while_ready
2024-06-02 00:15:35,396 - twister - DEBUG - DEVICE: PASS - test_ignore_while_ready in 0.007 seconds
2024-06-02 00:15:35,401 - twister - DEBUG - DEVICE: ===================================================================
2024-06-02 00:15:35,403 - twister - DEBUG - DEVICE: START - test_ignores
2024-06-02 00:15:35,408 - twister - DEBUG - DEVICE: PASS - test_ignores in 0.002 seconds
2024-06-02 00:15:35,414 - twister - DEBUG - DEVICE: ===================================================================
2024-06-02 00:15:35,418 - twister - DEBUG - DEVICE: TESTSUITE conn_mgr_monitor succeeded
2024-06-02 00:15:35,418 - twister - DEBUG - DEVICE: 
2024-06-02 00:15:35,421 - twister - DEBUG - DEVICE: ------ TESTSUITE SUMMARY START ------
2024-06-02 00:15:35,422 - twister - DEBUG - DEVICE: 
[... cut ...]
2024-06-02 00:15:35,533 - twister - DEBUG - DEVICE: - PASS - [conn_mgr_monitor.test_ignores] duration = 0.002 seconds
2024-06-02 00:15:35,533 - twister - DEBUG - DEVICE: 
2024-06-02 00:15:35,536 - twister - DEBUG - DEVICE: ------ TESTSUITE SUMMARY END ------
2024-06-02 00:15:35,537 - twister - DEBUG - DEVICE: 
2024-06-02 00:15:35,542 - twister - DEBUG - DEVICE: ===================================================================
2024-06-02 00:15:35,546 - twister - DEBUG - DEVICE: RunID: ebe979f38777ee389fa7578d0c5e730b
2024-06-02 00:15:35,548 - twister - DEBUG - DEVICE: PROJECT EXECUTION SUCCESSFUL
```

rlubos, 2024-06-20:
@hakehuang Try with the latest main, it's most likely the same issue that was reported [here](https://github.com/zephyrproject-rtos/zephyr/issues/74240), the fix was already merged: https://github.com/zephyrproject-rtos/zephyr/pull/74249

closed 2024-06-25 by hakehuang