zephyrproject-rtos/zephyr #74593: Missing WID errors

ranju28, 2024-06-20:
**Describe the bug**

I am observing missing WID errors in some stack conformance testcases.  I have executed with PTS 8.5.3. 
GAP/SEC/SEM/BV-28-C and GAP/SEC/SEM/BV-23-C  - No hdl_wid_550 found!
GAP/SEC/SEM/BI-13-C - No hdl_wid_353 found!
GAP/ADV/BV-20-C -    No hdl_wid_503 found!
GAP/BIS/BBM/BV-01-C - No hdl_wid_357 found!
GAP/SCN/BV-01-C - No hdl_wid_505 found!
L2CAP/ECFC/BV-43-C - No hdl_wid_140 found
How do we resolve this issue ? 

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using? [nRF52833](https://www.nordicsemi.com/Products/nRF52833)

**To Reproduce**
1. setup Zephyr environment and Auto PTS 
2. Build the auto pts tester app
3. Program the board
4. start the auto PTS server and Run the test cases GAP/SEC/SEM/BV-28-C GAP/SEC/SEM/BI-13-C GAP/ADV/BV-20-C GAP/BIS/BBM/BV-01-C GAP/SCN/BV-01-C L2CAP/ECFC/BV-43-C 

**Expected behavior**
All the test steps should be executed.

**Impact**
Unable to execute the above mentioned conformance tests. 

**Environment (please complete the following information):**
OS: Windows 10
Toolchain: Zephyr SDK
Zephyr Main

hermabe, 2024-06-20:
The "wid"s are implemented in the auto-pts repo and not in zephyr, so this issue does not belong here.

jori-nordic, 2024-06-20:
https://github.com/auto-pts/auto-pts/issues to add a link :)

jori-nordic, 2024-07-02:
Closing as per @hermabe 's reasoning

closed 2024-07-02 by jori-nordic