zephyrproject-rtos/zephyr #74594: zephyr kernel crash during networking testing on imx8mp Cortex-A Core

JiafeiPan, 2024-06-20:
**Describe the bug**

Kernel crash during running zperf on imx8mp Cortex-A Core.

**To Reproduce**
Steps to reproduce the behavior:
1. build Zephyr kernel: west build -p always -b imx8mp_evk/mimx8ml8/a53 samples/net/zperf/
2. Boot up imx8mp with networking cable plugged int.
3. Boot Zephyr on imx8mp Cortex-A Core:
    execute the following command in u-boot command line:
    tftp 0xC0000000 imx8mp/zephyr.bin;  dcache flush; icache flush; cpu 3  release 0xC0000000
4. on Zephyr Zperf console execute ping command
5. disconnect networking cable, then can find the following crash information:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
Zephyr kernel crash.

**Logs and console output**

*** Booting Zephyr OS build v3.7.0-rc1-129-g283204e984a4 ***
[00:00:00.027,000] <inf> net_config: Initializing network
[00:00:00.027,000] <inf> net_config: IPv4 address: 192.0.2.1
[00:00:00.127,000] <inf> net_config: IPv6 address: 2001:db8::1
[00:00:00.127,000] <inf> net_config: IPv6 address: 2001:db8::1
[00:00:03.032,000] <inf> phy_rt_rtl8211f: PHY 1 is up
[00:00:03.032,000] <inf> phy_rt_rtl8211f: PHY (1) Link speed 1000 Mb, full duplex
[00:00:03.032,000] <inf> eth_nxp_enet_mac: Link is up
uart:~$ net ping 192.0.2.2
PING 192.0.2.2
28 bytes from 192.0.2.2 to 192.0.2.1: icmp_seq=1 ttl=64 time=2.62 ms
28 bytes from 192.0.2.2 to 192.0.2.1: icmp_seq=2 ttl=64 time=1.50 ms
28 bytes from 192.0.2.2 to 192.0.2.1: icmp_seq=3 ttl=64 time=1.51 ms
uart:~$
[00:00:17.063,000] <inf> phy_rt_rtl8211f: PHY 1 is down
[00:00:17.063,000] <inf> eth_nxp_enet_mac: Link is down
[00:00:17.069,000] <err> os: ELR_ELn: 0x00000000c005e927
[00:00:17.069,000] <err> os: ESR_ELn: 0x000000008a000000
[00:00:17.069,000] <err> os:   EC:  0x22 (PC alignment fault exception.)
[00:00:17.069,000] <err> os:   IL:  0x1
[00:00:17.069,000] <err> os:   ISS: 0x0
[00:00:17.069,000] <err> os: FAR_ELn: 0x00000000c005e927
[00:00:17.069,000] <err> os: TPIDRRO: 0x01000000c005dcc8
[00:00:17.069,000] <err> os: x0:  0x0000000000000000  x1:  0x00000000c005ad60
[00:00:17.069,000] <err> os: x2:  0x00000000c0063040  x3:  0x00000000c005dcc8
[00:00:17.069,000] <err> os: x4:  0x00000000c0062f30  x5:  0x00000000ffffffff
[00:00:17.069,000] <err> os: x6:  0x00000000ffffffff  x7:  0x0000000000000040
[00:00:17.069,000] <err> os: x8:  0x0000000000000004  x9:  0x00000000c003f000
[00:00:17.069,000] <err> os: x10: 0x00000000c004b000  x11: 0x0000000000000002
[00:00:17.069,000] <err> os: x12: 0xaaaaaaaaaaaaaaaa  x13: 0xaaaaaaaaaaaaaaaa
[00:00:17.069,000] <err> os: x14: 0xaaaaaaaaaaaaaaaa  x15: 0x0000000000000000
[00:00:17.069,000] <err> os: x16: 0x0000000000000001  x17: 0x0000000000000000
[00:00:17.069,000] <err> os: x18: 0xaaaaaaaaaaaaaaaa  lr:  0x00000000c005e927
[00:00:17.069,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:17.069,000] <err> os: Current thread: 0xc0058ee0 (net_mgmt)
[00:00:17.272,000] <err> os: Halting system
uart:~$

dleach02, 2024-06-24:
@JiafeiPan can you bisect this to see if there was a particular commit that caused this.

JiafeiPan, 2024-06-25:
This issue has been fixed by PR: https://github.com/zephyrproject-rtos/zephyr/pull/74595, so I would like to close this issue, thanks.

closed 2024-06-25 by JiafeiPan