zephyrproject-rtos/zephyr #74601: native_sim: combining `-fstack-protector` with `CONFIG_POSIX_C_LIB_EXT=y` crashes during boot

mniestroj, 2024-06-20:
**Describe the bug**
Combining `CONFIG_POSIX_C_LIB_EXT=y` and `-fstack-protector` results in `native_sim` platform to crash during boot. For distributions that have -fstack-protector` as default (e.g. Arch Linux) it results in many integration tests and sample apps in Zephyr to fail (e.g. `west twister` does not pass).

**To Reproduce**
Steps to reproduce the behavior:
1. `west build -p -b native_sim zephyr/tests/posix/getentropy/ -T portability.posix.getentropy.picolibc -- -DEXTRA_CPPFLAGS=-fstack-protector` 
    - `-fstack-protector` is disabled by default on Ubuntu / https://hub.docker.com/r/zephyrprojectrtos/ci CI image
    - `-fstack-protector` is enabled by default on Arch Linux, so passing `-DEXTRA_CPPFLAGS=-fstack-protector` is not needed to reproduce this issue
2. `west build -t run`
3. `Segmentation fault (core dumped)` error occurs due to NULL pointer dereference
     - `__stack_chk_init` is called automatically by host libc, before even Zephyr init kicks in (since it is marked as `__attribute__((__constructor__))`
     - `getentropy()` is called, which is provided by Zephyr in case `CONFIG_POSIX_C_LIB_EXT=y`
     - `device_is_ready(entropy) == false`, so `errno = EIO` assignment is attempted
     - `errno = ...` is actually `_kernel.cpus[0].current.errno_var = ...` in that case; this results with segmentation fault since `_kernel.cpus[0].current == 0x0` at this stage

**Expected behavior**
Running `west build -p -b native_sim zephyr/tests/posix/getentropy/ -T portability.posix.getentropy.picolibc -t run` should pass, even on systems where `-fstack-protector` is the default (like on Arch Linux).

**Impact**
Applications that use `CONFIG_POSIX_C_LIB_EXT=y` (e.g. using thread-safe `getopt()` shell extension, crc shell command, `fnmatch()` function) fail to work with `native_sim`. Also "standard" Zephyr integration tests fail on Linux distributions with `-fstack-protector` being in the default host compiler flags.

**Logs and console output**

<details>
<summary>GDB session logs</summary>

```
Breakpoint 1, __stack_chk_init () at /modules/lib/picolibc/newlib/libc/ssp/stack_protector.c:23
23        if (__stack_chk_guard != 0)
(gdb) bt
#0  __stack_chk_init () at /modules/lib/picolibc/newlib/libc/ssp/stack_protector.c:23
#1  0xf7c52bc7 in call_init (argc=1, argv=0xffffa044, env=<optimized out>) at ../csu/libc-start.c:145
#2  __libc_start_main_impl (main=0x804936d <_start+45>, argc=1, argv=0xffffa044, init=0x0, fini=0x0, rtld_fini=0xf7fcdfb0 <_dl_fini>, stack_end=0xffffa03c)
    at ../csu/libc-start.c:347
#3  0x08049368 in _start ()
(gdb) f 1
#1  0xf7c52bc7 in call_init (argc=1, argv=0xffffa044, env=<optimized out>) at ../csu/libc-start.c:145
Downloading source file /usr/src/debug/glibc/glibc/csu/../csu/libc-start.c
145             ((dl_init_t) addrs[j]) (argc, argv, env);                                                                                                                          
(gdb) l
140         {
141           unsigned int jm
142             = l->l_info[DT_INIT_ARRAYSZ]->d_un.d_val / sizeof (ElfW(Addr));
143           ElfW(Addr) *addrs = (void *) (init_array->d_un.d_ptr + l->l_addr);
144           for (unsigned int j = 0; j < jm; ++j)
145             ((dl_init_t) addrs[j]) (argc, argv, env);
146         }
147     }
148
149     #else /* !SHARED */
(gdb) f 1
#1  0xf7c52bc7 in call_init (argc=1, argv=0xffffa044, env=<optimized out>) at ../csu/libc-start.c:145
145             ((dl_init_t) addrs[j]) (argc, argv, env);
(gdb) f 0
#0  __stack_chk_init () at /modules/lib/picolibc/newlib/libc/ssp/stack_protector.c:23
23        if (__stack_chk_guard != 0)
(gdb) s
26        if (getentropy) {
(gdb) 
28          getentropy(&__stack_chk_guard, sizeof(__stack_chk_guard));
(gdb) 
0x0804b1db in getentropy (buffer=0x805ad08 <__stack_chk_guard>, length=4) at /zephyr/lib/posix/options/getentropy.c:16
16      {
(gdb) 
24              if (length > 256) {
(gdb) 
29              if (entropy == NULL || !device_is_ready(entropy)) {
(gdb) p length
$1 = 4
(gdb) s
[... 50 lines cut ...]
</details>

<details>
<summary>`gcc -v` (Arch Linux)</summary>

```
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-pc-linux-gnu/14.1.1/lto-wrapper
Target: x86_64-pc-linux-gnu
Configured with: /build/gcc/src/gcc/configure --enable-languages=ada,c,c++,d,fortran,go,lto,m2,objc,obj-c++,rust --enable-bootstrap --prefix=/usr --libdir=/usr/lib --libexecdir=/usr/lib --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=https://gitlab.archlinux.org/archlinux/packaging/packages/gcc/-/issues --with-build-config=bootstrap-lto --with-linker-hash-style=gnu --with-system-zlib --enable-__cxa_atexit --enable-cet=auto --enable-checking=release --enable-clocale=gnu --enable-default-pie --enable-default-ssp --enable-gnu-indirect-function --enable-gnu-unique-object --enable-libstdcxx-backtrace --enable-link-serialization=1 --enable-linker-build-id --enable-lto --enable-multilib --enable-plugin --enable-shared --enable-threads=posix --disable-libssp --disable-libstdcxx-pch --disable-werror
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 14.1.1 20240522 (GCC)
```

</details>

**Environment (please complete the following information):**
 - OS: Linux, Arch Linux without explicit `-fstack-protector` flag
 - Toolchain: Zephyr SDK 0.16.8
 - Zephyr `3.7.0-rc1`

**Additional context**
Noticed that issue with downstream applications recently, after bumping to `3.7.0-rc1`. `git bisect` showed 2bc722a97ed6 ("posix: deprecate POSIX_FNMATCH GETOPT GETENTROPY") as the first failing commit. The reason was that `CONFIG_POSIX_C_LIB_EXT` is now enabled for more applications than `CONFIG_GETENTROPY` was. As a result downstream app with `getopt()` use is now failing to boot with `native_sim`, due to `getentropy()` being implemented by Zephyr POSIX subsystem (which wasn't the case before, since `CONFIG_GETENTROPY=n` before Zephyr version bump).

mniestroj, 2024-06-20:
FYI @keith-packard @cfriedt

cfriedt, 2024-06-20:
It might be possible to fix the issue in native_sim by setting the default cpu to a "dummy" cpu at build time to avoid the null pointer dereference.

Would that solve the issue for you @mniestroj ?

mniestroj, 2024-06-20:
If you mean:
```
modified   kernel/init.c
@@ -44,7 +44,9 @@ BUILD_ASSERT(CONFIG_MP_NUM_CPUS == CONFIG_MP_MAX_NUM_CPUS,
 
 /* the only struct z_kernel instance */
 __pinned_bss
-struct z_kernel _kernel;
+struct z_kernel _kernel = {
+	.cpus[0].current = &_thread_dummy,
+};
 
 #ifdef CONFIG_PM
 __pinned_bss atomic_t _cpus_active;
 ```
 then yes, it helps to fix/workaround the crash.

mniestroj, 2024-06-20:
> @mniestroj I pressume that if you pass `-DEXTRA_CPPFLAGS=-fno-stack-protector` in your Arch build the issue goes away, correct? (I don't have Arch here to test)

@aescolar Yes, that is right.

aescolar, 2024-06-20:
> @aescolar Yes, that is right.

Thanks. The issue would seem to be that we are getting the picolibC/embedded stack protector code being used. Which I very much thing we should not.
A quick "fix" may be to just say that we do not support enabling the stack-protector feature when building with picolibc for the posix arch, which I doubt anybody will miss.
That could mean an if in the posix arch cmake file to set -fno-stack-protector if the compiler supports it.
Anyhow, I'm looking into this. Thanks for the report :)

PR #74682 (2024-06-25): Run functions with __attribute__((__constructor__)) on init
For the motivation please check  #74659, but in very short: initializer functions (with `__attribute__((__constructor__))` are not exclusive to C++, but are also used by some libraries. Zephyr not supporting them without C++ means those libraries will not work as intended. This PR fixes this by allowing running those initializers also without enabling C++.

* Move ctors and init_array initialization from the CPP library to the kernel library, as this is common for both C and C++ and it is the kernel who is running it.
* Rename z_cpp_i [... cut]
closed 2024-06-25 by nashif