zephyrproject-rtos/zephyr #74607: Drivers adc.h not taking oversampling from the devicetree in the resolution calculation

Robibobo1, 2024-06-20:
**Bug Description**

When using the `adc_raw_to_millivolts_dt` function at line 804 of the `include/zephyr/drivers/adc.h` file, the oversampling described in the devicetree is not applied. This results in an incorrect millivolt value.

I discovered this bug by debugging line by line and analyzing when the value became incorrect.

I am using the SAMD21 family, and to correct the problem, I added these lines of code:

```c
/* Adjust resolution for oversampling */
if(spec->oversampling) {
    resolution += spec->oversampling >> 1U;
}
```

**Note:** This method of correcting the resolution works for this family because of the following table in the datasheet:

[image]

As you can see, an oversampling of 2 adds 1 bit of resolution, 4 adds 2, etc. That’s why I shifted one bit to the right to divide it by 2.

I have no idea if this will work with other microchip families.

decsny, 2024-07-05:
This really does not seem like a generic problem, retitled and changed priority to low

guenzel-kinexon, 2024-08-07:
The proposed "fix" is actually not solving the actual issue.
The SAMD21 ADC has an internal dividing mechanism which is partially automatic and partially manually to be set.

In detail: The `ADJRES` field in the `AVGCTRL` register has to be set according to the datasheet.

```c
// adc_sam0.c:329
adc->AVGCTRL.reg = ADC_AVGCTRL_SAMPLENUM(sequence->oversampling);
switch (sequence->oversampling){
	case 0:
	case 1:
	case 2:
	case 3:
		adc->AVGCTRL.reg |= ADC_AVGCTRL_ADJRES(sequence->oversampling);	
		break;
	case 4:
	default:
		adc->AVGCTRL.reg |= ADC_AVGCTRL_ADJRES(4);		
}
```

PR #76963 (2025-01-15): drivers: adc: SAM0: Enable automatic result devision when oversampling
The SAM0 ADC has an internal register to divide the accumulated results of oversampling the ADC by the amount of samples collected. This value has to be set by the driver and was missing.

Fixes  #74607
closed 2025-01-15 by kartben