zephyrproject-rtos/zephyr #74623: POSIX sysconf breaks builds for many STM32 targets

besmarsh, 2024-06-20:
**Describe the bug**

Zephyr v3.6 added a compile-time `sysconf()` implementation to the POSIX API, enabled with the Kconfig option `CONFIG_POSIX_SYSCONF`. One of the options that can be queried is `PAGESIZE`/`PAGE_SIZE` (`_SC_PAGESIZE`/`_SC_PAGE_SIZE`). Enabling `CONFIG_POSIX_SYSCONF` and calling `sysconf(_SC_PAGESIZE)` fails to compile for many STM32 targets. This is obviously an issue for applications wishing to call `sysconf(_SC_PAGESIZE)` on an affected target.

Zephyr v3.7 added a runtime `sysconf()` implementation, deprecated `CONFIG_POSIX_SYSCONF`, and added `CONFIG_POSIX_SYSCONF_IMPL_MACRO` and `CONFIG_POSIX_SYSCONF_IMPL_FULL` to select between the compile-time and runtime `sysconf()` implementations. When `CONFIG_POSIX_SYSCONF_IMPL_MACRO=y`, the behaviour is the same as in v3.6. When `CONFIG_POSIX_SYSCONF_IMPL_FULL=y`, the problem is worse as applications will fail to compile even if they don't call `sysconf()` at all. Even worse, `CONFIG_POSIX_SYSCONF_IMPL_FULL` is the default when `CONFIG_POSIX_API=y` and `CONFIG_CPP=y`, meaning that even applications not using sysconf at all will fail to compile if they enable the POSIX API (now needed for POSIX net socket names after the deprecation of `CONFIG_NET_SOCKETS_POSIX_NAMES`) and C++ support.

The bug is caused by:

* Sysconf options are identified by `__z_posix_sysconf*` macros (defined in `unistd.h` in v3.6, and `sysconf.h` in v3.7)
* `__z_posix_sysconf_SC_PAGE_SIZE` is defined as `PAGE_SIZE`, which comes from `CONFIG_POSIX_PAGE_SIZE_BITS` in v3.6 and `CONFIG_POSIX_PAGE_SIZE` in v3.7
* `__z_posix_sysconf_SC_PAGESIZE` is defined as `PAGESIZE`, which is defined as `PAGE_SIZE` (as above), IF `PAGESIZE` is not already defined
* `PAGESIZE` is defined in the STM32 HAL file `stm32_hal_legacy.h`, which eventually gets included from `unistd.h`, as `FLASH_PAGE_SIZE`
*  `FLASH_PAGE_SIZE` is defined in the STM32 HAL file `stm32**xx_hal_flash.h` for some but not all STM32 families

Applications targeting STM32 families that have `FLASH_PAGE_SIZE` defined in their  `stm32**xx_hal_flash.h` will compile, whereas applications targeting STM32 families without `FLASH_PAGE_SIZE` defined will not.

This raises another issue - for those STM32 families with `FLASH_PAGE_SIZE` defined where applications using `sysconf()` do compile, the results of `sysconf(_SC_PAGESIZE)` and `sysconf(_SC_PAGE_SIZE)` may be different, despite the documentation for `sysconf` describing `PAGE_SIZE`/`_SC_PAGE_SIZE` as "a synonym for `PAGESIZE`/`_SC_PAGESIZE`.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
   - Will not compile for stm32h735g_disco, among other targets
 - What have you tried to diagnose or workaround this issue?
   - Diagnosis described above
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
   - Issue is not present prior to v3.6 as `sysconf()` was not implemented
   - Issue is present in v3.6 but only affects applications using `sysconf()`
   - Issue is present in v3.7 and affects both applications using `sysconf()`, and applications not using `sysconf()` but using POSIX API and C++

**To Reproduce**
Steps to reproduce the behavior:
1. `cd zephyrproject`
2. `cmake -b stm32h735g_disco -d build_sysconf_bug zephyr/samples/hello_world -- -DCONFIG_POSIX_API=y -DCONFIG_CPP=y`
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

- Applications not even using `sysconf` should compile
- Applications using `sysconf` should compile
- Calling `sysconf(_SC_PAGESIZE)` and `sysconf(_SC_PAGE_SIZE)` should yield the same result

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

For affected targets:

- Applications not using sysconf: Not a showstopper as they can set `CONFIG_POSIX_SYSCONF_IMPL_MACRO=y` to avoid the issue, but it takes some investigation to discover this workaround
- Appliations using compile-time sysconf: Annoyance as they can use `_SC_PAGE_SIZE` rather than `_SC_PAGESIZE`
- Application using runtime sysconf: Showstopper

**Logs and console output**
```
In file included from .../zephyrproject/modules/hal/stm32/stm32cube/stm32h7xx/drivers/include/stm32h7xx_hal_def.h:30,
                 from .../zephyrproject/modules/hal/stm32/stm32cube/stm32h7xx/drivers/include/stm32h7xx_hal_rcc.h:27,
                 from .../zephyrproject/modules/hal/stm32/stm32cube/stm32h7xx/drivers/include/stm32h7xx_hal_conf.h:246,
                 from .../zephyrproject/modules/hal/stm32/stm32cube/stm32h7xx/drivers/include/stm32h7xx_hal.h:29,
                 from .../zephyrproject/modules/hal/stm32/stm32cube/stm32h7xx/soc/stm32h7xx.h:282,
                 from c:\users\benma\zephyrproject\zephyr\soc\st\stm32\stm32h7x\soc.h:12,
                 from c:\users\benma\zephyrproject\zephyr\modules\cmsis\cmsis_core_m.h:24,
                 from c:\users\benma\zephyrproject\zephyr\modules\cmsis\cmsis_core.h:10,
                 from .../zephyrproject/zephyr/include/zephyr/arch/arm/asm_inline_gcc.h:24,
                 from .../zephyrproject/zephyr/include/zephyr/arch/arm/asm_inline.h:18,
                 from .../zephyrproject/zephyr/include/zephyr/arch/arm/arch.h:33,
                 from .../zephyrproject/zephyr/include/zephyr/arch/cpu.h:19,
                 from .../zephyrproject/zephyr/include/zephyr/kernel_includes.h:36,
                 from .../zephyrproject/zephyr/include/zephyr/kernel.h:17,
                 from .../zephyrproject/zephyr/include/zephyr/posix/pthread.h:13,
                 from .../zephyrproject/zephyr/lib/posix/options/sysconf.c:7:
[... 32 lines cut ...]
  105 | #define UTIL_PRIMITIVE_CAT(a, ...) a##__VA_ARGS__
      |                                    ^
.../zephyrproject/zephyr/include/zephyr/sys/util_internal.h:104:26: note: in expansion of macro 'UTIL_PRIMITIVE_CAT'
  104 | #define UTIL_CAT(a, ...) UTIL_PRIMITIVE_CAT(a, __VA_ARGS__)
      |                          ^~~~~~~~~~~~~~~~~~
.../zephyrproject/zephyr/include/zephyr/sys/util_internal.h:120:29: note: in expansion of macro 'UTIL_CAT'
  120 | #define _CONCAT_1(arg, ...) UTIL_CAT(arg, _CONCAT_0(__VA_ARGS__))
      |                             ^~~~~~~~
.../zephyrproject/zephyr/include/zephyr/sys/util_internal.h:105:36: note: in expansion of macro '_CONCAT_1'
  105 | #define UTIL_PRIMITIVE_CAT(a, ...) a##__VA_ARGS__
      |                                    ^
.../zephyrproject/zephyr/lib/posix/options/sysconf.c:13:28: note: in expansion of macro 'CONCAT'
   13 | #define z_sysconf(x) (long)CONCAT(__z_posix_sysconf, x)
      |                            ^~~~~~
.../zephyrproject/zephyr/lib/posix/options/sysconf.c:241:24: note: in expansion of macro 'z_sysconf'
  241 |                 return z_sysconf(_SC_PAGESIZE);
      |                        ^~~~~~~~~
[115/173] Building C object zephyr/lib/posix/options/CMakeFiles/lib__posix__options.dir/pthread.c.obj
ninja: build stopped: subcommand failed.
```

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain: Zephyr SDK (arm-zephyr-eabi)
 - Commit SHA or Version used: v3.6, `main` at time of investigation (`7f0f0a4`)

besmarsh, 2024-06-20:
I've just seen #70136, #73363, and #74428 regarding this same issue.

erwango, 2024-06-24:
@cfriedt Should we keep this open now that https://github.com/zephyrproject-rtos/zephyr/pull/74428 was merged.
At least maybe the STM32 label can be removed if #74624 is still considered (as this one is generic).

cfriedt, 2024-06-24:
@erwango - I think we can close this. The other pr is less of a bug fix and more of a stability improvement, I would say.