zephyrproject-rtos/zephyr #74628: Cellular modem with Sara-R5 turning off fail

jeronimoagullo, 2024-06-20:
I am using the `modem_cellular` driver with MQTT to connect to Azure IoTHub. Everything works smoothly, but when I try to turning off the modem it fails always with a **-11 (EAGAIN)** error.

**Description of my environment**
Currently I am using the latest v3.7.0-rc1 Zephyr version with a micromod/nrf52840 board and an ublox Sara-R5 modem.

**Describe the bug**
The modem can resolve DNS and connect to MQTT (Azure) broker correctly. Then, after a time, the device enters into "sleep mode", the following code is used to enter into "sleep mode" carrying out the following steps:
1. Disconnect from mqtt broker
2. Suspend the modem
3. Suspend the modem UART
4. Entering into "power off"

```c
static void sleep_mode_entering(struct k_work *work)
{
        const struct device *modem = DEVICE_DT_GET(DT_ALIAS(modem));

	// disconnect mqtt
	ret = mqtt_disconnect(&client_ctx);
	if (ret < 0){
		LOG_ERR("Error during MQTT disconnection, %d", ret);
		return;
	}

	mqtt_connected = false;
	LOG_INF("MQTT client disconnecting...");

	// wait until MQTT client is disconnected
	k_sem_take(&mqtt_stop, K_FOREVER);

	// Turn off modem
	ret = pm_device_action_run(modem, PM_DEVICE_ACTION_SUSPEND);
	if (ret < 0){
		LOG_ERR("Error turning off the modem, %d", ret);
		return;
	}

        // Turn off UART
	ret = pm_device_action_run(uart, PM_DEVICE_ACTION_SUSPEND);
	if (ret < 0){
		LOG_ERR("Error turning off the uart, %d", ret);
		return;
	}

	// Entering SoC into sleep. Waking up only wakeup_pin by reset
	sys_poweroff();

	return;
}
```

However, the result of the `pm_device_action_run(modem, PM_DEVICE_ACTION_SUSPEND);` function is always **-11 (EAGAIN)** error, that is, a time out taking the semaphore `data->suspended_sem` within the `modem_cellular_pm_action` function.

**Expected behavior**
Turning off the modem through the `power_gpio` pin when calling the function `pm_device_action_run(modem, PM_DEVICE_ACTION_SUSPEND);`

**Additional context**
I have checked the `modem_cellular` code and it seems that the semaphore `data->suspended_sem` is used to prevent entering into suspend state if the modem is not in the **idle** state. Actually, when turning off the modem, the current state is `MODEM_CELLULAR_STATE_CARRIER_ON`.

Thus, I wonder how can I enter into idle state before turning off the modem or what should be the correct steps to powering off the modem. Thank you in advance!

aescolar, 2024-06-25:
CC @bjarki-andreasen

bjarki-andreasen, 2024-06-25:
Hi, this is likely because you are calling the sync pm commands from the work queue thread :) The modem subsystem including the cellular modem device driver run in the system work queue using k_work items. Since the pm_device_action_run() blocks the thread that calls it, the cellular modem driver is deadlocked :)

Try calling suspend from another thread, main for example :) or create a dedicated work queue for the application to use :)