zephyrproject-rtos/zephyr #74635: BT tests/bsim/bluetooth/ll/multiple_id/tests_scripts/multiple.sh failing in main for nrf54l15bsim/nrf54l15/cpuap

aescolar, 2024-06-20:
**Describe the bug**
tests/bsim/bluetooth/ll/multiple_id/tests_scripts/multiple.sh is failing in main when run for the nrf54l15bsim/nrf54l15/cpuap

**To Reproduce**
Steps to reproduce the behavior:
1. nice tests/bsim/bluetooth/compile.nrf54l15bsim_nrf54l15_cpuapp.sh 
2. BOARD=nrf54l15bsim/nrf54l15/cpuapp tests/bsim/bluetooth/ll/multiple_id/tests_scripts/multiple.sh
3. See error

**Expected behavior**
No test failure

**Impact**
CI blocked for BT related PRs

**Logs and console output**

**Environment (please complete the following information):**
 - OS: Ubuntu 22.04
 - Toolchain (e.g Zephyr SDK, ...): Host gcc 12
 - Commit SHA or Version used: main as of now b38d4300b9711c48fb62d37c2536e28004ef1858

**Additional context**
The issue started being visible in main after 94d712e5cfe001996d7280c028e0e0303cf7b747, but the underlaying issue seems to be the controller LLL using uninitialized variables.

cvinayak, 2024-06-21:
> ==51336== ERROR SUMMARY: 10000000 errors from 152 contexts (suppressed: 0 from 0)
==51336== 
==51336== 1 errors in context 1 of 152:
==51336== Conditional jump or move depends on uninitialised value(s)
==51336==    at 0x80979CD: ticker_enqueue.isra.0 (ticker.c:577)
==51336==    by 0x8097BBC: ticker_job_insert.isra.0 (ticker.c:2355)
==51336==    by 0x80994C1: ticker_job_list_insert (ticker.c:2836)
==51336==    by 0x809A8FB: ticker_job (ticker.c:3181)
==51336==    by 0x809518D: mayfly_run (mayfly.c:217)
==51336==    by 0x80CA4E9: rtc0_nrf5_isr (lll.c:137)
==51336==    by 0x80619EC: vector_to_irq (irq_handler.c:64)
==51336==    by 0x8061C9B: posix_irq_handler (irq_handler.c:119)
==51336==    by 0x8061545: posix_halt_cpu (soc.c:99)
==51336==    by 0x8060CC6: arch_cpu_idle (cpuhalt.c:40)
==51336==    by 0x80E943D: k_cpu_idle (kernel.h:5918)
==51336==    by 0x80E9583: idle (idle.c:74)
==51336==  Uninitialised value was created by a stack allocation
==51336==    at 0x80A8BC1: ticker_update_rand.constprop.0.isra.0 (ull_adv.c:1944)
==51336== 
==51336== 
==51336== 1 errors in context 2 of 152:
==51336== Conditional jump or move depends on uninitialised value(s)
==51336==    at 0x8097972: ticker_enqueue.isra.0 (ticker.c:586)
==51336==    by 0x8097BBC: ticker_job_insert.isra.0 (ticker.c:2355)
==51336==    by 0x80993B0: ticker_job_list_insert (ticker.c:2836)
==51336==    by 0x809A8FB: ticker_job (ticker.c:3181)
==51336==    by 0x809518D: mayfly_run (mayfly.c:217)
==51336==    by 0x80CA4E9: rtc0_nrf5_isr (lll.c:137)
==51336==    by 0x80619EC: vector_to_irq (irq_handler.c:64)
==51336==    by 0x8061C9B: posix_irq_handler (irq_handler.c:119)
==51336==    by 0x8061545: posix_halt_cpu (soc.c:99)
==51336==    by 0x8060CC6: arch_cpu_idle (cpuhalt.c:40)
==51336==    by 0x80E943D: k_cpu_idle (kernel.h:5918)
==51336==    by 0x80E9583: idle (idle.c:74)
==51336==  Uninitialised value was created by a stack allocation
==51336==    at 0x80A8BC1: ticker_update_rand.constprop.0.isra.0 (ull_adv.c:1944)
==51336== 
...

The errors reported by valgrind are related to missing random number generator (controller implementation uses uninitiliazed stack memory to return random numbers in the preliminary port), currently random number are used from stack allocation as also stated by valgrind. 

Lowering the priority as the test is not failing on the latest main e1c51f9a722, valgrind does have errors, but test pass.

aescolar, 2024-06-21:
Thanks Vinayak. Please leave the priority as high until the test disabling is merged. The test does fail. But main's push CI job (the one you linked) does not run bsim tests.

aescolar, 2024-08-24:
The test is still disabled => removing stale

aescolar, 2024-10-24:
The controller for a nRF54L15  is not anymore just pulling random stack content as random values (at least for the simulated target) So the issue is not present anymore.
=> Closing

closed 2024-10-24 by aescolar