zephyrproject-rtos/zephyr #74647: #include "../../../../../../../media/mi/5e58d875-5e34-6b43-8936-d956a2f60c04/works/github/zephyrproject/zephyr/arch/arm/core/vector_table.ld"       |                                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ compilation terminated. [79/315] Building C object zephyr/arch/arch/arm/core/cortex_m/CMakeFi...hub__zephyrproject__zephyr__arch__arm__core__cortex_m.dir/fault.c.obj ninja: build stopped: subcommand failed.

LingaoM, 2024-06-21:
If you modify the configuration through menuconfig, an error will be reported when compiling again.

For example:

1. west build -b bbc_microbit_v2 zephyr/tests/bluetooth/shell/ -p auto  
2. west build -t menuconfig, type `/` input `BT_DEBUG_MONITOR_UART` and enter save.
3. west build

modules/hal/nordic/nrfx/mdk -I/home/mi/works/github/zephyrproject/zephyr/modules/hal_nordic/nrfx/. -I/media/mi/5e58d875-5e34-6b43-8936-d956a2f60c04/works/github/zephyrproject/modules/hal/st/sensor/stmemsc/lis2mdl_STdC/driver/ -I/media/mi/5e58d875-5e34-6b43-8936-d956a2f60c04/works/github/zephyrproject/modules/crypto/tinycrypt/lib/include -I/home/mi/works/github/zephyrproject/zephyr/include/zephyr/arch/arm/cortex_m/scripts -D__GCC_LINKER_CMD__ -DLINKER_ZEPHYR_PREBUILT -E /home/mi/works/github/zephyrproject/zephyr/include/zephyr/arch/arm/cortex_m/scripts/linker.ld -P -o linker_zephyr_pre0.cmd && /usr/bin/cmake -E cmake_transform_depfile Ninja gccdepfile /home/mi/works/github/zephyrproject/zephyr/tests/bluetooth/shell /home/mi/works/github/zephyrproject/zephyr /home/mi/works/github/zephyrproject/build /home/mi/works/github/zephyrproject/build/zephyr /home/mi/works/github/zephyrproject/build/zephyr/linker_zephyr_pre0.cmd.dep /home/mi/works/github/zephyrproject/build/CMakeFiles/d/af1ffabef38d460b6c582d0d863c22e37f606b7bb6feae921eb21593aefd1607.d
In file included from /home/mi/works/github/zephyrproject/zephyr/include/zephyr/arch/arm/cortex_m/scripts/linker.ld:145:
/home/mi/works/github/zephyrproject/build/zephyr/include/generated/snippets-rom-start.ld:4:38: fatal error: ../../../../../../../media/mi/5e58d875-5e34-6b43-8936-d956a2f60c04/works/github/zephyrproject/zephyr/arch/arm/core/vector_table.ld: No such file or directory
    4 | /* Sort key: "0x0vectors" */#include "../../../../../../../media/mi/5e58d875-5e34-6b43-8936-d956a2f60c04/works/github/zephyrproject/zephyr/arch/arm/core/vector_table.ld"
      |                                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
[23/237] Building C object zephyr/lib/libc/common/CMakeFiles/lib__libc__common.dir/source/stdlib/malloc.c.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /media/mi/5e58d875-5e34-6b43-8936-d956a2f60c04/works/github/zephyrproject/build

carlescufi, 2024-06-21:
I cannot reproduce this @LingaoM. Just tried the exact instructions locally but could not reproduce it.

ithinuel, 2024-06-21:
You seem to be using two instances of zephyr, one in /media and one in /home. Is that intended ?

LingaoM, 2024-06-21:
> You seem to be using two instances of zephyr, one in /media and one in /home. Is that intended ?

yeh, media is zephyr home is ncs.

carlescufi, 2024-06-21:
@LingaoM can you confirm that you can reproduce this?

LingaoM, 2024-06-21:
> can you confirm that you can reproduce this?

sure

carlescufi, 2024-06-21:
> > You seem to be using two instances of zephyr, one in /media and one in /home. Is that intended ?
> 
> yeh, media is zephyr home is ncs.

It's a bit strange that they are both present in the command-line invocation? Are you sure this is not a local issue on your side?

LingaoM, 2024-06-22:
I guess the problem cause by i'm install zephyr in separate disk, while i'm create **ln -s** into /home/works.

LingaoM, 2024-06-22:
lrwxrwxrwx  1 mi   mi          53 Jul 25  2023  works -> /media/mi/5e58d875-5e34-6b43-8936-d956a2f60c04/works/

LingaoM, 2024-06-22:
[image]

marc-hb, 2024-06-27:
> I guess the problem cause by i'm install zephyr in separate disk, while i'm create ln -s into /home/works.

Symbolic links are a bit of a hack. Among others, symbolic links very often break ".."

```shell
mkdir -p top/subdir/
ln -s top/subdir/ subsym
cd subsym

ls .. # where does this go? It depends on the language/environment, etc.
```

- https://lwn.net/Articles/899543/
- https://github.com/zephyrproject-rtos/west/issues/419
- #50284 
- etc.

Zephyr is not completely incompatible with symbolic links but having the workspace spread across multiple disks joined by a symbolic link is really asking for trouble.

Please try not to stitch a workspace with symbolic links? Also, please provide complete reproduction instructions that someone else can use.

LingaoM, 2024-06-27:
> Zephyr is not completely incompatible with symbolic links but having the workspace spread across multiple disks joined by a symbolic link is really asking for trouble.
>
> Please try not to stitch a [...]

If you explicitly ask Zephyr not to support workspace with symbolic links, then you should state it in the documentation, otherwise, someone will always take this step.

LingaoM, 2024-06-27:
> Also, please provide complete reproduction instructions that someone else can use.

Since Zephyr & toochain is quite large, and the user space of my Linux default partition is already very compact, but my SSD path is mounted to /media/mi/5e58d875-5e34-6b43-8936-d956a2f60c04/, this path is quite long, so I created a soft link to access this SSD in the `~` directory:
``` C
lrwxrwxrwx  1 mi   mi          53 Jul 25  2023  works -> /media/mi/5e58d875-5e34-6b43-8936-d956a2f60c04/works/
```

Then I installed Zephyr in the `~/works` directory, so Zephyr is actually in the SSD.
Then https://github.com/zephyrproject-rtos/zephyr/issues/74647#issue-2365971330

marc-hb, 2024-06-27:
> If you explicitly ask Zephyr not to support workspace with symbolic links, then you should state it in the documentation,

There are plenty of developers who compile Zephyr while using symbolic links. If you skim the links I posted above you will quickly see that supporting symbolic links is not a "yes/no" question. The answer is always "it depends" because you can create arbitrarily complicated situations with symbolic links. So it's literally impossible to document, it's always on a case by case basis. Again, check the past examples I posted.

> Since Zephyr & toochain is quite large

Does "Zephyr" mean "Zephyr workspace"?

Does "toolchain" mean "Zephyr SDK"?

> and the user space of my Linux default partition is already very compact,

Does "user space" mean "$HOME"?

What does "default partition" mean? I don't know this concept.

Does "compact" mean "nearly full"?

> but my SSD path is mounted to...

Do you mean _second_ SSD? 

This is all very hard to follow. Please use standard Zephyr and Linux terminology and step-by-step instructions. Imagine that you are guiding someone trying to replicate your setup - because you are!

> so I created a soft link to access this SSD in the ~ directory
> works -> /media/mi/5e58d875-5e34-6b43-8936-d956a2f60c04/works/

Thanks, that part is very specific. More like this please.

LingaoM, 2024-06-27:
> Does "Zephyr" mean "Zephyr workspace"?

Y

> Does "toolchain" mean "Zephyr SDK"?

Y

> Does "user space" mean "$HOME"?

Y

> What does "default partition" mean? I don't know this concept.

Linux system default partition

> Does "user space" mean "$HOME"?

Y

> Do you mean second SSD?

Y

closed 2024-07-05 by LingaoM