zephyrproject-rtos/zephyr #74659: Zephyr does not run functions with __attribute__((__constructor__)) on init unless CONFIG_CPP=y

aescolar, 2024-06-21:
This is a set of 2 very related bugs

## 1st. Zephyr does not run functions with `__attribute__((__constructor__))` on init unless CONFIG_CPP=y

**Describe the bug**
Unless CONFIG_CPP is set, Zephyr does not execute during init functions with `__attribute__((__constructor__)) `
(i.e. functions in .ctors or .init_array sections) or manually placed in this sections.
This is not correct behavior, as this is not CPP exclusive, and some libraries rely this behavior. For example
https://github.com/picolibc/picolibc/blob/main/newlib/libc/ssp/stack_protector.c#L20 

On the other hand if CONFIG_CPP is set, this will be correctly called together with C++ object constructors.

**To Reproduce**
Steps to reproduce the behavior:
Modify hello world like so:
```diff
diff --git a/samples/hello_world/src/main.c b/samples/hello_world/src/main.c
index c550ab461cb..c4994a4aba9 100644
--- a/samples/hello_world/src/main.c
+++ b/samples/hello_world/src/main.c
@@ -5,6 +5,13 @@
  */
 
 #include <stdio.h>
+#include <string.h>
+
+__attribute__((__constructor__))
+void foobar(void)
+{
+       printf("hello from embedded function with attribute constructor\n");
+}
 
 int main(void)
 {
```
1. mkdir build; cd build
2. cmake -GNinja -DBOARD=qemu_x86 ../samples/hello_world/
3. ninja run
4. See there is no "hello from embedded function with attribute constructor" printed.
```
SeaBIOS (version zephyr-v1.0.0-0-g31d4e0e-dirty-20200714_234759-fv-az50-zephyr)
Booting from ROM..
*** Booting Zephyr OS build v3.7.0-rc1-196-g5e295ac39ebd ***
Hello World! qemu_x86/atom
```

1. mkdir build; cd build
2. cmake -GNinja -DBOARD=qemu_x86 ../samples/hello_world/ -DCONFIG_CPP=y
3. ninja run
4. See "hello from embedded function with attribute constructor" is printed.
```
SeaBIOS (version zephyr-v1.0.0-0-g31d4e0e-dirty-20200714_234759-fv-az50-zephyr)
Booting from ROM..
*** Booting Zephyr OS build v3.7.0-rc1-196-g5e295ac39ebd ***
hello from embedded function with attribute constructor  <--------------
Hello World! qemu_x86/atom
```

**Expected behavior**
Functions w `__attribute__((__constructor__)) ` (in  .ctors or .init_array sections) called even if CONFIG_CPP is not set

**Impact**
Some applications/libraries will not initialize properly

**Environment (please complete the following information):**
 - OS: Linux, Ubuntu 22.04
 - Toolchain (e.g Zephyr SDK, ...): 0.16.8
 - Commit SHA or Version used: main as of now: 5e295ac39ebd7a732cc064cdf72433f16392b021

## 2nd. Functions w `__attribute__((__constructor__))` in native_sim are run before Zephyr is booted unless CONFIG_CPP=y

Due to the issues described above, as those .ctors and .init_array sections are forgotten by Zephyr's linker script unless CONFIG_CPP=y, the final link of the native_simulator runner will pick them. On runtime the host libC, while initializing the native_simulator runner, will call into them.
This will call this embedded functions to be executed in the context of the runner, before anything related to Zephyr is initialized , which is likely to cause trouble.
If on the other hand CONFIG_CPP=y is set for the Zephyr image, they will be properly called during Zephyr boot and only then.

**To Reproduce**
For extra context, we modify the sample a bit more:
```diff
diff --git a/samples/hello_world/src/runnery.c b/samples/hello_world/src/runnery.c
new file mode 100644
index 00000000000..4b2e13ac551
--- /dev/null
+++ b/samples/hello_world/src/runnery.c
@@ -0,0 +1,7 @@
+#include <stdio.h>
+
+__attribute__((__constructor__))
+void foobar_runner (void)
+{
+       printf("hello from host/runner context function with attribute constructor\n");
+}
diff --git a/samples/hello_world/CMakeLists.txt b/samples/hello_world/CMakeLists.txt
index ecb7d24bb8f..6c26da39b6b 100644
--- a/samples/hello_world/CMakeLists.txt
+++ b/samples/hello_world/CMakeLists.txt
@@ -6,3 +6,5 @@ find_package(Zephyr REQUIRED HINTS $ENV{ZEPHYR_BASE})
 project(hello_world)
 
 target_sources(app PRIVATE src/main.c)
+
+target_sources(native_simulator INTERFACE src/runnery.c)
```

1.cmake -GNinja -DBOARD=native_sim ../samples/hello_world/ 
2.ninja && ./zephyr/zephyr.exe
```
hello from embedded function with attribute constructor
hello from host/runner context function with attribute constructor
*** Booting Zephyr OS build v3.7.0-rc1-196-g5e295ac39ebd ***
Hello World! native_sim/native
```
3. See that "hello from embedded function with attribute constructor" appears before Zephyr's boot and next to hello from host/runner context function with attribute constructor

On the other hand
1.cmake -GNinja -DBOARD=native_sim ../samples/hello_world/ 
2.ninja && ./zephyr/zephyr.exe
```
hello from host/runner context function with attribute constructor
*** Booting Zephyr OS build v3.7.0-rc1-197-g66ef30d8c633 ***
hello from embedded function with attribute constructor
Hello World! native_sim/native
```
3. Properly runs the embedded constructor at Zephyr boot.

**Additional context**
Related to #74601 (which is a particular bug due to this behaviour)
Related to #39347 , #36907, #46357, 8e35cc8eb4ec10f50819fa402dbc3c44b6bd25ca which added or improved the handling of these sections

aescolar, 2024-06-21:
Given that I seem to be the first one to realize (or at least report this) I will set it as low priority.
But this should at least prevent picolibc's stack-protector from working on embedded targets. Apart from being the underlaying caused the issue described here #74601

aescolar, 2024-06-21:
CC @mniestroj  @keith-packard

keith-packard, 2024-06-21:
This used to work.

andyross, 2024-06-21:
FWIW, the code to call those is in lib/cpp/abi and predicated on CONFIG_CPP_STATIC_INIT_GNU=y.  My guess is if @keith-packard saw it happen it was because there's some code in the libc that's duplicating it.

It shouldn't be hard to unwrap that and make it universal; maybe just move it into somewhere near the top of z_cstart() in kernel/init.c...

aescolar, 2024-06-21:
> This used to work.

@keith-packard what I can see is that for embedded targets, the stack-protector code is called properly except for the init function, and still detects and reports the failures.

keith-packard, 2024-06-21:
Yeah, when I dug in I noticed that only Zephyr constructors using SYS_INIT were run from C applications. We've got two nearly identical solutions in #74682 and #74700; both of those will increase ROM usage for all Zephyr C applications.

PR #74682 (2024-06-25): Run functions with __attribute__((__constructor__)) on init
For the motivation please check  #74659, but in very short: initializer functions (with `__attribute__((__constructor__))` are not exclusive to C++, but are also used by some libraries. Zephyr not supporting them without C++ means those libraries will not work as intended. This PR fixes this by allowing running those initializers also without enabling C++.

* Move ctors and init_array initialization from the CPP library to the kernel library, as this is common for both C and C++ and it is the kernel who is running it.
* Rename z_cpp_i [... cut]
closed 2024-06-25 by nashif