zephyrproject-rtos/zephyr #74662: Port `usb_dc_smartbond` to `USB_DEVICE_STACK_NEXT`

carlescufi, 2024-06-21:
In order for the `USB_DEVICE_STACK_NEXT` to become the default, existing USB device drivers must be ported to it.
See #42066 for additional details.

carlescufi, 2024-08-26:
@ioannis-karachalios are you going to post a PR porting this USB driver?

carlescufi, 2024-08-27:
@sjanc do you know if there's any ongoing work related to this?

sjanc, 2024-08-28:
@carlescufi we are likely going to start working on this this week

ioannis-karachalios, 2024-09-02:
@carlescufi, thanks for the info provided and sorry for my late response. We will assign an engineer to start working on USB porting and most likely we will be able to kick it off mid this week.

kasjer, 2024-09-12:
@carlescufi I'm currently working on this port, is there a way for me to assign this issue to myself?

carlescufi, 2024-09-18:
> @carlescufi I'm currently working on this port, is there a way for me to assign this issue to myself?

I see @henrikbrixandersen did that, thanks @kasjer.

carlescufi, 2024-09-24:
@kasjer do you have an ETA for a PR?

kasjer, 2024-09-24:
@carlescufi I expect to finish it in a week or so.

PR #79184 (2024-10-22): drivers: usb: udc: add Smartbond UDC driver
Code adds USB UDC driver for Smartbond device.

Code was NOT tested with ISO endpoints yet.
It was tested with console, mass, and cdc_acm applications

Additionally small fix in DMA power management was needed.

Function dma_smartbond_set_channel_status() used incorrect
condition to release state lock.
In initialization function dma_smartbond_init() function
dma_smartbond_set_channel_status() was called for each DMA
channel and tried to release lock that was never taken.

Fixes #74662
closed 2024-10-22 by henrikbrixandersen