zephyrproject-rtos/zephyr #74665: Port `usb_dc_sam_usbc` to `USB_DEVICE_STACK_NEXT`

carlescufi, 2024-06-21:
In order for the `USB_DEVICE_STACK_NEXT` to become the default, existing USB device drivers must be ported to it.
See #42066 for additional details.

Note that if this driver is not ported by Zephyr 4.0.0, this hardware will not work with the default USB stack in 4.0.0.

carlescufi, 2024-08-27:
@nandojve do you have plans to port this yourself?

nandojve, 2024-08-27:
> @nandojve do you have plans to port this yourself?

Hi @carlescufi ,

I only use my free time to work on Zephyr and, at moment, I'm quite busy. So, if anyone have time to do it it is Ok from my side. I can help testing on real HW at least. I run Linux loopback driver and report.