zephyrproject-rtos/zephyr #74675: Enabling the CONFIG_ESP32_WIFI_NET_ALLOC_SPIRAM does not move the WiFi noinit to the SPIRAM

marekmatej, 2024-06-21:
**Describe the bug**
Enabling the CONFIG_ESP32_WIFI_NET_ALLOC_SPIRAM=y should move the `.noinit` sections to the SPIRAM and therefore freeing the SRAM memory. This is not the case

Memory listing without the `CONFIG_ESP32_WIFI_NET_ALLOC_SPIRAM=y`:
```
Memory region         Used Size  Region Size  %age Used
           FLASH:      623856 B    4194048 B     14.87%
     iram0_0_seg:       84224 B       235 KB     35.00%
     dram0_0_seg:      121864 B       192 KB     61.98%
```
Memory listing with the `CONFIG_ESP32_WIFI_NET_ALLOC_SPIRAM=y`:
```
Memory region         Used Size  Region Size  %age Used
           FLASH:      624796 B    4194048 B     14.90%
     iram0_0_seg:       89344 B       235 KB     37.13%
     dram0_0_seg:      123528 B       192 KB     62.83%
     ...
     ext_ram_seg:      262134 B         2 MB     12.50%
```

**To Reproduce**
Build the wifi sample and observe the available SRAM:
`west build -b esp32s3_devkitm/esp32s3/procpu  samples/net/wifi/ -p -DCONFIG_ESP32_WIFI_NET_ALLOC_SPIRAM=y -DCONFIG_ESP_SPIRAM=y`

**Expected behavior**
There is no more available SRAM memory after enabling the option.

**Impact**
The option does not provide the promised SRAM memory gain.

**Fix**
https://github.com/zephyrproject-rtos/zephyr/pull/74670

EricNRS, 2024-08-22:
This is still an important issue that needs to be resolved.  Please keep this open.

KevinJordil, 2024-12-10:
Is anyone still working on it?

marekmatej, 2024-12-10:
> Is anyone still working on it?

Yep