zephyrproject-rtos/zephyr #74713: [Coverity CID: 392520] Same on both sides in arch/riscv/core/stacktrace.c

zephyrbot, 2024-06-22:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/dcf42917c550714d2457947538b9e29d083e872e/arch/riscv/core/stacktrace.c

Category: Incorrect expression
Function: `walk_stackframe`
Component: Architectures
CID: [392520](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=392520)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/dcf42917c550714d2457947538b9e29d083e872e/arch/riscv/core/stacktrace.c#L191

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=392520

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_CONSTANT_EXPRESSION_RESULT). ([CWE-569](http://cwe.mitre.org/data/definitions/569.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

ycsin, 2024-06-22:
Can't tell what's wrong with the LOC. Could this be a false positive?

nashif, 2024-07-11:
[image]

PR #77067 (2024-08-26): arch: reorg stacktrace dependencies & add `arch_stack_walk()` for x86 & arm64
> [!NOTE]
> Expected to conflict with #77457

Fixes #76706
Fixes #74713

Includes #76044 & #76042, obsoletes #75865

___

1. If an arch implements `arch_stack_walk()` `ARCH_HAS_STACKWALK` should be selected.
2. If the above is enabled, `ARCH_SUPPORTS_STACKWALK` indicates if the dependencies are met for arch to enable stack walking. This Kconfig replaces `<arch>_EXCEPTION_STACK_TRACE`
2. If the above is enabled, then, `ARCH_STACKWALK` determines if `arch_stack_walk()` should be compiled.
3. `EXCEPTION_STACK_TRACE` sho [... cut]
closed 2024-08-26 by nashif