zephyrproject-rtos/zephyr #74718: [Coverity CID: 330735] Uninitialized scalar variable in subsys/bluetooth/controller/ll_sw/ull_filter.c

zephyrbot, 2024-06-22:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/dcf42917c550714d2457947538b9e29d083e872e/subsys/bluetooth/controller/ll_sw/ull_filter.c

Category: Uninitialized variables
Function: `rpa_adv_refresh`
Component: Bluetooth
CID: [330735](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=330735)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/dcf42917c550714d2457947538b9e29d083e872e/subsys/bluetooth/controller/ll_sw/ull_filter.c#L1236

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=330735

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_UNINIT). ([CWE-457](http://cwe.mitre.org/data/definitions/457.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

aescolar, 2024-07-10:
Closely related to https://github.com/zephyrproject-rtos/zephyr/issues/74716

aescolar, 2024-07-11:
@cvinayak any update on these?

aescolar, 2024-07-12:
Lowering priority. This code has run with valgrind quite a times and nothing was reported so it is either not happening often enough or a false positive. More analysis is needed though.

closed 2024-11-23 by kartben