zephyrproject-rtos/zephyr #74726: [Coverity CID: 368785] Constant expression result in subsys/bluetooth/host/classic/hfp_ag.c

zephyrbot, 2024-06-22:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/dcf42917c550714d2457947538b9e29d083e872e/subsys/bluetooth/host/classic/hfp_ag.c

Category: Integer handling issues
Function: `bt_hfp_ag_remote_terminate`
Component: Bluetooth
CID: [368785](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=368785)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/dcf42917c550714d2457947538b9e29d083e872e/subsys/bluetooth/host/classic/hfp_ag.c#L2244

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=368785

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_CONSTANT_EXPRESSION_RESULT). ([CWE-569](http://cwe.mitre.org/data/definitions/569.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

lylezhu2012, 2024-06-24:
@jhedberg ,  I cannot access the link https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=368785. Could you guide me how to register an account.

jhedberg, 2024-06-24:
> @jhedberg , I cannot access the link https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=368785. Could you guide me how to register an account?

@lylezhu2012 you need to use the "sign in with saml" and then "sign in with GitHub" links. This should trigger a request to the admins (AFAIK). I think @aescolar @carlescufi or @nashif should have privileges to grant you access.

commit 59981228 (2024-06-27): Bluetooth: Classic: HF_AG: Fix incorrect status judgment

Wrong condition of AG status is used for status
checking.

Fixes #74726.
closed 2024-06-27 by aescolar