zephyrproject-rtos/zephyr #74729: [Coverity CID: 392500] Dereference null return value in subsys/bluetooth/audio/cap_initiator.c

zephyrbot, 2024-06-22:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/dcf42917c550714d2457947538b9e29d083e872e/subsys/bluetooth/audio/cap_initiator.c

Category: Null pointer dereferences
Function: `bt_cap_initiator_unicast_audio_update`
Component: Bluetooth
CID: [392500](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=392500)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/dcf42917c550714d2457947538b9e29d083e872e/subsys/bluetooth/audio/cap_initiator.c#L1478

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=392500

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_NULL_RETURNS). ([CWE-476](http://cwe.mitre.org/data/definitions/476.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

babrsn, 2024-06-24:
@Thalley please assign it to @babrsn (if nobody else has started working on it)

babrsn, 2024-06-25:
@Thalley since issues #74739 and #74733 are very small and somehow related to this issue, maybe they can all be assigned to me and be fixed in one PR..

Thalley, 2024-06-25:
> @Thalley since issues #74739 and #74733 are very small and somehow related to this issue, maybe they can all be assigned to me and be fixed in one PR..

The 2 CAP initiator issue makes sense to have in the same PR, but please do create separate PRs for unrelated changes (such as https://github.com/zephyrproject-rtos/zephyr/issues/74733).

commit 0c611d01 (2024-06-27): Bluetooth: Audio: Dereference null return value in cap_initiator.c

Null pointer dereferences in functions:
  bt_cap_initiator_unicast_audio_update (Line 1478)
  bt_cap_initiator_unicast_audio_stop (Line 1693)

Fixes #74729 and #74739
closed 2024-06-27 by nashif