zephyrproject-rtos/zephyr #74731: [Coverity CID: 392502] Out-of-bounds access in subsys/bluetooth/host/gatt.c

zephyrbot, 2024-06-22:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/dcf42917c550714d2457947538b9e29d083e872e/subsys/bluetooth/host/gatt.c

Category: Memory - corruptions
Function: `db_hash_finish`
Component: Bluetooth
CID: [392502](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=392502)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/dcf42917c550714d2457947538b9e29d083e872e/subsys/bluetooth/host/gatt.c#L777

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=392502

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_OVERRUN). ([CWE-119](http://cwe.mitre.org/data/definitions/119.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

jhedberg, 2024-06-24:
@ceolin same thing here, with Coverity apparently using different array sizes for analyzing the library and analyzing the Zephyr main tree. I.e. this looks also like a false positive.

alwa-nordic, 2024-06-27:
@nashif

How does Coverity model this code?

 - Does it "build" a specific application? Which one? Which config?
 - Which toolchain does it model?

My best understanding is that Coverity essentially looks at compile_commands.json. But, compile_commands.json will contain multiple compilation units. Is it possible to see which compilation unit, including the GCC equivalent command line, is used in a particular report?

nashif, 2024-06-27:
@alwa-nordic 

please do not remove priority, you can change periority, but do not remove it. If it is a false positive, mark this in the tool.

coverity wraps around twister  and running all available tests for a select set of platforms, it does not use the json file.

closed 2024-07-01 by zephyrbot