zephyrproject-rtos/zephyr #74732: [Coverity CID: 392506] Dereference after null check in subsys/bluetooth/host/classic/a2dp.c

zephyrbot, 2024-06-22:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/dcf42917c550714d2457947538b9e29d083e872e/subsys/bluetooth/host/classic/a2dp.c

Category: Null pointer dereferences
Function: `bt_a2dp_stream_start`
Component: Bluetooth
CID: [392506](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=392506)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/dcf42917c550714d2457947538b9e29d083e872e/subsys/bluetooth/host/classic/a2dp.c#L752

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=392506

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_FORWARD_NULL). ([CWE-476](http://cwe.mitre.org/data/definitions/476.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

jhedberg, 2024-07-03:
@lylezhu2012 any update here?

lylezhu2012, 2024-07-03:
> @lylezhu2012 any update here?

PR is created for it.

PR #75368 (2024-07-05): Bluetooth: A2DP: Fix NULL pointer references issue
The opposite logic is used to determine if a pointer is valid.

Correct the judgement logic.

Fixes #74732.
commit 6fad6585 (2024-07-05): Bluetooth: A2DP: Fix NULL pointer references issue

The opposite logic is used to determine if a
pointer is valid.

Correct the judgment logic.

Fixes #74732.
closed 2024-07-05 by aescolar