zephyrproject-rtos/zephyr #74734: [Coverity CID: 392517] Logically dead code in subsys/bluetooth/host/classic/a2dp.c

zephyrbot, 2024-06-22:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/dcf42917c550714d2457947538b9e29d083e872e/subsys/bluetooth/host/classic/a2dp.c

Category: Control flow issues
Function: `a2dp_set_config_ind`
Component: Bluetooth
CID: [392517](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=392517)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/dcf42917c550714d2457947538b9e29d083e872e/subsys/bluetooth/host/classic/a2dp.c#L200

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=392517

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_DEADCODE). ([CWE-561](http://cwe.mitre.org/data/definitions/561.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

jhedberg, 2024-07-03:
@lylezhu2012 any update here?

lylezhu2012, 2024-07-03:
> @lylezhu2012 any update here?

PR is created for it.

PR #75369 (2024-07-05): Bluetooth: A2DP: Check the pointer before using
Check the pointer `sep` before using it.

Fixes #74734.
commit c9708ff9 (2024-07-05): Bluetooth: A2DP: Check the pointer before using

Check the pointer `sep` before using it.

Fixes #74734.
closed 2024-07-05 by aescolar