zephyrproject-rtos/zephyr #74738: [Coverity CID: 392529] Dereference after null check in subsys/bluetooth/controller/ll_sw/ull_adv_sync.c

zephyrbot, 2024-06-22:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/dcf42917c550714d2457947538b9e29d083e872e/subsys/bluetooth/controller/ll_sw/ull_adv_sync.c#L1386

Category: Null pointer dereferences
Function: `ull_adv_sync_copy_pdu_header`
Component: Bluetooth
CID: [392529](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=392529)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/dcf42917c550714d2457947538b9e29d083e872e/subsys/bluetooth/controller/ll_sw/ull_adv_sync.c

```
1357                             source_dptr += sizeof(struct pdu_cte_info);
1358                     }
1359     #endif /* CONFIG_BT_CTLR_DF_ADV_CTE_TX */
1360    
1361     #if defined(CONFIG_BT_CTLR_ADV_PERIODIC_ADI_SUPPORT)
1362                     if (source_hdr->ext_hdr.adi) {
>>>     CID 392529:    (FORWARD_NULL)
>>>     Dereferencing null pointer "skip_fields".
1363                             if (!skip_fields->adi) {
1364                                     memcpy(target_dptr, source_dptr, sizeof(struct pdu_adv_adi));
1365                                     target_dptr += sizeof(struct pdu_adv_adi);
1366                                     target_hdr->ext_hdr.adi = 1U;
1367                             }
1368                             source_dptr += sizeof(struct pdu_adv_adi);
1380                     }
1381     #endif /* CONFIG_BT_CTLR_ADV_SYNC_PDU_LINK */
1382    
1383                     /* SyncInfo is RFU for periodic advertising */
1384    
1385                     if (source_hdr->ext_hdr.tx_pwr) {
>>>     CID 392529:    (FORWARD_NULL)
>>>     Dereferencing null pointer "skip_fields".
1386                             if (!skip_fields->tx_pwr) {
1387                                     *target_dptr = *source_dptr;
1388                                     target_dptr++;
1389                                     target_hdr->ext_hdr.tx_pwr = 1U;
1390                             }
1391                             source_dptr++;
```

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_FORWARD_NULL). ([CWE-476](http://cwe.mitre.org/data/definitions/476.html))

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=392529

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

ljd42, 2025-07-09:
Analysis:
------------
The function `ull_adv_sync_copy_pdu_header` is declared as static, and therefore only visible within that compilation unit. 
It is always called with a non-null `skip_fields` pointer to a struct pdu_adv_ext_hdr:
```
$ grep -rn "ull_adv_sync_copy_pdu_header" subsys/bluetooth
subsys/bluetooth/controller/ll_sw/ull_adv_sync.c:1368:static void ull_adv_sync_copy_pdu_header(struct pdu_adv *target_pdu,
subsys/bluetooth/controller/ll_sw/ull_adv_sync.c:1636:			ull_adv_sync_copy_pdu_header(pdu_chain, pdu, &skip_fields, true);
subsys/bluetooth/controller/ll_sw/ull_adv_sync.c:1771:		ull_adv_sync_copy_pdu_header(ter_pdu, ter_pdu_prev, &skip_fields, false);
subsys/bluetooth/controller/ll_sw/ull_adv_sync.c:2265:		ull_adv_sync_copy_pdu_header(pdu_chain, pdu, &skip_fields, true);
```
Therefore, the reported  "Dereferencing null pointer "skip_fields" is a False Positive the way  `ull_adv_sync_copy_pdu_header` is currently called.

However it has a potential bug, should we pass in the future a `NULL skip_fields`.

ljd42, 2025-07-09:
Didn't realize there was already a PR for this. Got unfortunately not merged and closed?

PR #92920 (2025-07-19): Bluetooth: Controller: Fix a potential NULL pointer dereference
Fix a potential NULL pointer dereference for skip_fields in the function ull_adv_sync_copy_pdu_header. This issue been reported by Coverity.

Coverity CID: 392529
Fixes: #74738
closed 2025-07-19 by nashif