zephyrproject-rtos/zephyr #74739: [Coverity CID: 392533] Dereference null return value in subsys/bluetooth/audio/cap_initiator.c

zephyrbot, 2024-06-22:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/dcf42917c550714d2457947538b9e29d083e872e/subsys/bluetooth/audio/cap_initiator.c#L1693

Category: Null pointer dereferences
Function: `bt_cap_initiator_unicast_audio_stop`
Component: Bluetooth
CID: [392533](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=392533)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/dcf42917c550714d2457947538b9e29d083e872e/subsys/bluetooth/audio/cap_initiator.c#L1693

```
1687    
1688             /** TODO: If this is a CSIP set, then the order of the procedures may
1689              * not match the order in the parameters, and the CSIP ordered access
1690              * procedure should be used.
1691              */
1692             proc_param = get_next_proc_param(active_proc);
>>>     CID 392533:  Null pointer dereferences  (NULL_RETURNS)
>>>     Dereferencing "proc_param", which is known to be "NULL".
1693             bap_stream = &proc_param->stream->bap_stream;
1694             active_proc->proc_initiated_cnt++;
1695    
1696             err = bt_bap_stream_release(bap_stream);
1697             if (err != 0) {
1698                     LOG_DBG("Failed to stop bap_stream %p: %d", proc_param->stream, err);
```

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_NULL_RETURNS). ([CWE-476](http://cwe.mitre.org/data/definitions/476.html))

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=392533

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

Thalley, 2024-06-26:
@babrsn please comment here so I can assign you to this issue :)

babrsn, 2024-06-26:
@Thalley please assign it to @babrsn

closed 2024-06-27 by nashif