zephyrproject-rtos/zephyr #74740: [Coverity CID: 392536] Uninitialized pointer read in subsys/bluetooth/host/classic/a2dp.c

zephyrbot, 2024-06-22:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/dcf42917c550714d2457947538b9e29d083e872e/subsys/bluetooth/host/classic/a2dp.c#L258

Category: Memory - illegal accesses
Function: `a2dp_set_config_ind`
Component: Bluetooth
CID: [392536](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=392536)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/dcf42917c550714d2457947538b9e29d083e872e/subsys/bluetooth/host/classic/a2dp.c#L258

```
252                      stream->remote_ep = NULL;
253                      stream->codec_config = *cfg.codec_config;
254                      ep->stream = stream;
255              }
256      }
257    
>>>     CID 392536:  Memory - illegal accesses  (UNINIT)
>>>     Using uninitialized value "stream".
258      ops = stream->ops;
259      if (*errcode == 0) {
260              if ((ops != NULL) && (stream->ops->configured != NULL)) {
261                      stream->ops->configured(stream);
262              }
263      }
```

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_UNINIT). ([CWE-457](http://cwe.mitre.org/data/definitions/457.html))

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=392536

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

jhedberg, 2024-07-03:
@lylezhu2012 any update here?

lylezhu2012, 2024-07-03:
> @lylezhu2012 any update here?

Yes. Fixed.

PR #75366 (2024-07-05): Bluetooth: A2DP: Check if pointer is valid
Only access the pointer `stream` if it is valid.

Fixes #74735 
Fixes #74740
commit 9be685ee (2024-07-05): Bluetooth: A2DP: Access pointer if it is valid

Only access if the pointer `stream` is valid.

Fixes #74735
Fixes #74740
closed 2024-07-05 by aescolar