zephyrproject-rtos/zephyr #74745: [Coverity CID: 348195] Out-of-bounds access in drivers/gnss/gnss_nmea_generic.c

zephyrbot, 2024-06-22:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/dcf42917c550714d2457947538b9e29d083e872e/drivers/gnss/gnss_nmea_generic.c

Category: Memory - corruptions
Function: `gnss_nmea_generic_init_chat`
Component: Drivers
CID: [348195](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=348195)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/dcf42917c550714d2457947538b9e29d083e872e/drivers/gnss/gnss_nmea_generic.c#L138

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=348195

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_OVERRUN). ([CWE-119](http://cwe.mitre.org/data/definitions/119.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

bjarki-andreasen, 2024-06-25:
I believe this is a false positive, same false positive as https://github.com/zephyrproject-rtos/zephyr/issues/69115

@aescolar do you agree?

aescolar, 2024-06-26:
Same as https://github.com/zephyrproject-rtos/zephyr/issues/74746#issuecomment-2191663888

bjarki-andreasen, 2024-06-26:
Marked as false positive in Coverity

closed 2024-06-26 by bjarki-andreasen