zephyrproject-rtos/zephyr #74746: [Coverity CID: 348212] Out-of-bounds access in drivers/gnss/gnss_nmea_generic.c

zephyrbot, 2024-06-22:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/dcf42917c550714d2457947538b9e29d083e872e/drivers/gnss/gnss_nmea_generic.c

Category: Memory - corruptions
Function: `gnss_nmea_generic_init_pipe`
Component: Drivers
CID: [348212](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=348212)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/dcf42917c550714d2457947538b9e29d083e872e/drivers/gnss/gnss_nmea_generic.c#L115

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=348212

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_OVERRUN). ([CWE-119](http://cwe.mitre.org/data/definitions/119.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

bjarki-andreasen, 2024-06-25:
I believe this is a false positive, same false positive as #69115

@aescolar do you agree?

aescolar, 2024-06-26:
> @aescolar do you agree?

I think so. The way to proceed would be for you to set in coverity the "clasification" , "action" and "owner" (yourself) and write a comment justifying the conclusion on it being a false positive, and then closing this bug in Zephyr.

bjarki-andreasen, 2024-06-26:
@aescolar I think I need more privileges to do so, the `Triage` options like `Classification` are greyed out for me :)

aescolar, 2024-06-26:
> @aescolar I think I need more privileges to do so, the `Triage` options like `Classification` are greyed out for me :)

try now

bjarki-andreasen, 2024-06-26:
It worked, marked as false positive

closed 2024-06-26 by bjarki-andreasen