zephyrproject-rtos/zephyr #74749: [Coverity CID: 349571] Macro compares unsigned to 0 in drivers/sensor/ens160/ens160.c

zephyrbot, 2024-06-22:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/dcf42917c550714d2457947538b9e29d083e872e/drivers/sensor/ens160/ens160.c

Category: Integer handling issues
Function: `ens160_set_humidity`
Component: Drivers
CID: [349571](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=349571)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/dcf42917c550714d2457947538b9e29d083e872e/drivers/sensor/ens160/ens160.c#L65

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=349571

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_NO_EFFECT). ([CWE-570](http://cwe.mitre.org/data/definitions/570.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

ubieda, 2024-07-03:
I left a review in Coverity:
> This redundant check is intentional. The comparison lower than zero is done inside the macro DIV_ROUND_CLOSEST and I believe it doesn't make sense to further optimize it.

Classified as: Intentional.
Severity: Minor.
Action: Ignore.

ubieda, 2024-07-03:
Closing this issue. Feel free to re-open if I'm missing anything.

closed 2024-07-03 by ubieda