zephyrproject-rtos/zephyr #74751: [Coverity CID: 353647] Operands don't affect result in drivers/mfd/mfd_axp192.c

zephyrbot, 2024-06-22:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/dcf42917c550714d2457947538b9e29d083e872e/drivers/mfd/mfd_axp192.c#L555

Category: Integer handling issues
Function: `mfd_axp192_gpio_read_port`
Component: Drivers
CID: [353647](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=353647)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/dcf42917c550714d2457947538b9e29d083e872e/drivers/mfd/mfd_axp192.c#L555

```
549      gpio_input_val |=
550              (((gpio34_val & AXP192_GPIO34_INTPUT_MASK) >> AXP192_GPIO34_INTPUT_SHIFT) << 3u);
551    
552      gpio_output_val = (gpio012_val & AXP192_GPIO012_OUTPUT_MASK);
553      gpio_output_val |= ((gpio34_val & AXP192_GPIO34_OUTPUT_MASK) << 3u);
554      gpio_output_val |=
>>>     CID 353647:  Integer handling issues  (CONSTANT_EXPRESSION_RESULT)
>>>     "(gpio5_val & 4U) >> 3U" is 0 regardless of the values of its operands. This occurs as a value.
555              (((gpio5_val & AXP192_GPIO5_OUTPUT_MASK) >> AXP192_GPIO5_OUTPUT_SHIFT) << 5u);
556    
557      *value = gpio_input_val & ~(data->gpio_mask_output);
558      *value |= (gpio_output_val & data->gpio_mask_output);
559    
560      return 0;
```

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_CONSTANT_EXPRESSION_RESULT). ([CWE-569](http://cwe.mitre.org/data/definitions/569.html))

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=353647

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

aescolar, 2024-06-25:
CC @MrMarteng  @gmarull @aasinclair

MrMarteng, 2024-07-14:
@aescolar Thanks. I am on it. I identified, that there was a general mixup with GPIO5.

ljd42, 2025-07-05:
Following the AXP192-X Datasheet, the register 10h EXTEN & DC-DC2 output-control has the following layout: 

| Bit | Description                        | R/W | Default |
|------|-------------------------------------|--------| :--------: |
| 7-3 | reserved                             |          |              |
|  2    | EXTEN Switch control   <br />   0: shut down; 1: turn on |  RW  |      X      |
|  1    | reserved                            |          |               | 
|  0    | DC-DC2 Switch control   <br />  0: shut down; 1: turn on |  RW  |      x      |

If "GPIO5" corresponds to EXTEN Switch control (AXP192_GPIO5_OUTPUT_MASK is 0x04U), then the shift AXP192_GPIO5_OUTPUT_SHIFT should be 2, not 3.

kartben, 2025-07-08:
@MrMarteng would be great if you happen to find cycles to look into it :)

closed 2025-10-04 by cfriedt