zephyrproject-rtos/zephyr #74757: [Coverity CID: 363724] Out-of-bounds access in drivers/ethernet/nxp_enet/eth_nxp_enet.c

zephyrbot, 2024-06-22:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/dcf42917c550714d2457947538b9e29d083e872e/drivers/ethernet/nxp_enet/eth_nxp_enet.c

Category: Memory - corruptions
Function: `eth_nxp_enet_init`
Component: Drivers
CID: [363724](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=363724)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/dcf42917c550714d2457947538b9e29d083e872e/drivers/ethernet/nxp_enet/eth_nxp_enet.c#L737

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=363724

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_ARRAY_VS_SINGLETON). ([CWE-119](http://cwe.mitre.org/data/definitions/119.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

dleach02, 2024-06-25:
I snippet with additional information:

[image]

dleach02, 2024-06-25:
This is a false positive. Coverity is complaining that the pointer to buffer_config is being accessed as an array which is true. But the array size is somewhat hidden from coverity's scanner in the enet_config structure which is initialized to 1 in the call above this to ENET_GetDefaultConfig(). The buffer_config can be an array.

But @decsny is going to submit a PR to make it more clear to the reader that this is an array of 1. Not sure if coverity will still flag this given the somewhat hidden nature of the array size.

PR #74978 (2024-06-26): drivers: nxp_enet: Fix CID 392512 and CID 363724
The first commit fixes coverity issue 392512.
The issue was that the return value of clock_control_on was not being checked. Fix by checking the return value.

Fixes #74778

The second commit addresses the coverity issue ID 363724.
This issue seems to have been a false positive, but the root complaint
from coverity was that within the callstack of ENET_Up, some functions
use the buffer config pointer as an array, whereas in the zephyr driver
it is declared as a singleton. Address it by redeclaring this as an
array of size 1 instead of a singleto [... cut]
closed 2024-06-26 by nashif