zephyrproject-rtos/zephyr #74758: [Coverity CID: 363727] Operands don't affect result in drivers/mfd/mfd_ad559x_spi.c

zephyrbot, 2024-06-22:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/dcf42917c550714d2457947538b9e29d083e872e/drivers/mfd/mfd_ad559x_spi.c

Category: Integer handling issues
Function: `mfd_ad559x_spi_read_reg`
Component: Drivers
CID: [363727](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=363727)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/dcf42917c550714d2457947538b9e29d083e872e/drivers/mfd/mfd_ad559x_spi.c#L50

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=363727

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_CONSTANT_EXPRESSION_RESULT). ([CWE-569](http://cwe.mitre.org/data/definitions/569.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

aescolar, 2024-06-25:
CC @jeppenodgaard @gmarull @aasinclair

ljd42, 2025-07-15:
@aasinclair , @gmarull or @jeppenodgaard : could one of you quickly review the previous analysis? Thanks!

ljd42, 2025-09-27:
Marked as intentional in Coverity.

closed 2025-09-27 by ljd42