zephyrproject-rtos/zephyr #74762: [Coverity CID: 366275] Untrusted array index read in drivers/sensor/grow_r502a/grow_r502a.c

zephyrbot, 2024-06-22:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/dcf42917c550714d2457947538b9e29d083e872e/drivers/sensor/grow_r502a/grow_r502a.c

Category: Insecure data handling
Function: `r502a_validate_rx_packet`
Component: Drivers
CID: [366275](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=366275)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/dcf42917c550714d2457947538b9e29d083e872e/drivers/sensor/grow_r502a/grow_r502a.c#L106

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=366275

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_TAINTED_SCALAR). ([CWE-129](http://cwe.mitre.org/data/definitions/129.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

MaureenHelm, 2024-06-24:
@DineshDK03 please take a look

fabocode, 2025-01-23:
Hey @MaureenHelm I'm interested in contributing to Zephyr. I want to help with this specific issue.

MaureenHelm, 2025-01-24:
> Hey [[image]@MaureenHelm](https://github.com/MaureenHelm) I'm interested in contributing to Zephyr. I want to help with this specific issue.

Thanks @fabocode !

PR #84461 (2025-01-30): sensor: grow_r502a: fix: untrusted array index read
- Fixed Coverity CID 366275 by adding bounds check on packet length.

Fixes #74762
closed 2025-01-30 by kartben