zephyrproject-rtos/zephyr #74767: [Coverity CID: 368788] Use of 32-bit time_t in drivers/net/nsos_adapt.c

zephyrbot, 2024-06-22:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/dcf42917c550714d2457947538b9e29d083e872e/drivers/net/nsos_adapt.c

Category: High impact quality
Function: `nsos_adapt_setsockopt`
Component: Drivers
CID: [368788](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=368788)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/dcf42917c550714d2457947538b9e29d083e872e/drivers/net/nsos_adapt.c

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=368788

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_Y2K38_SAFETY). ([CWE-197](http://cwe.mitre.org/data/definitions/197.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

nashif, 2024-07-02:
@mniestroj  can you please take a look?

mniestroj, 2024-07-03:
Not sure what the issue really is here. Is there a chance to get more context or be able to login to Coverity for the details if this issue?

nashif, 2024-07-03:
[image]

This is probably related to https://github.com/zephyrproject-rtos/zephyr/issues/69109

mniestroj, 2024-07-03:
Since we are using host (Linux) side `setsockopt(..., SO_SNDTIMEO, ...)`, we are not able to do anything with how `time_t` is defined and what arguments (`struct timeval` which wraps `time_t`) are passed here to `setsockopt()`.

I think we should mark that particular issue as false-positive.

aescolar, 2024-07-08:
A couple of comments:
* Given that this is only one driver for testing in a test platform and a year 2039 issue, I don't think it is reasonable to classify it as high. So I will lower it to medium.
* We can (at least for glibc) have time_t be a 64bit value, and I have been thinking of doing the move soon to avoid the 2039Y issue. But even if we do so, I wonder if Coverity will know we are doing that. I would not do that before 3.7 though, as I fear possible issues (glibc added support for 64bit time_t in 2.34)

aescolar, 2024-07-08:
Here how it could be done https://github.com/zephyrproject-rtos/zephyr/pull/75567

carlescufi, 2024-07-31:
@mniestroj @aescolar any progress?

closed 2024-07-31 by aescolar