zephyrproject-rtos/zephyr #74805: [Coverity CID: 392526] Dereference null return value in subsys/net/lib/ptp/transport.c

zephyrbot, 2024-06-22:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/dcf42917c550714d2457947538b9e29d083e872e/subsys/net/lib/ptp/transport.c#L327

Category: Null pointer dereferences
Function: `ptp_transport_protocol_addr`
Component: Networking
CID: [392526](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=392526)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/dcf42917c550714d2457947538b9e29d083e872e/subsys/net/lib/ptp/transport.c#L327

```
321      int length = 0;
322    
323      if (IS_ENABLED(CONFIG_PTP_UDP_IPv4_PROTOCOL)) {
324              struct in_addr *ip = net_if_ipv4_get_global_addr(port->iface, NET_ADDR_PREFERRED);
325    
326              length = NET_IPV4_ADDR_SIZE;
>>>     CID 392526:  Null pointer dereferences  (NULL_RETURNS)
>>>     Dereferencing "ip", which is known to be "NULL".
327              *addr = ip->s_addr;
328      } else if (IS_ENABLED(CONFIG_PTP_UDP_IPv6_PROTOCOL)) {
329              struct in6_addr *ip = net_if_ipv6_get_global_addr(NET_ADDR_PREFERRED, &port->iface);
330    
331              length = NET_IPV6_ADDR_SIZE;
332              memcpy(addr, ip, length);
```

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_NULL_RETURNS). ([CWE-476](http://cwe.mitre.org/data/definitions/476.html))

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=392526

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

carlescufi, 2024-12-04:
@awojasinski please take another look at Coverity, still complains after your fix.

carlescufi, 2025-01-08:
Fixed by #83158

PR #74945 (2024-06-27): net: lib: ptp: Fix issues raised by static analyzer
Fix minor issues reported by static analysis tool:
- #74799
- #74800 
- #74801
- #74804 
- #74805 
- #74806 
- #74807 
- #74808 
- #74809
- #74811 
- #74812

Fixes #74799
Fixes #74800 
Fixes #74801
Fixes #74804 
Fixes #74805 
Fixes #74806 
Fixes #74807 
Fixes #74808 
Fixes #74809
Fixes #74811 
Fixes #74812
closed 2025-01-08 by carlescufi