zephyrproject-rtos/zephyr #74806: [Coverity CID: 392530] Unintentional integer overflow in subsys/net/lib/ptp/port.c

zephyrbot, 2024-06-22:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/dcf42917c550714d2457947538b9e29d083e872e/subsys/net/lib/ptp/port.c#L91

Category: Integer handling issues
Function: `port_timer_set_timeout`
Component: Networking
CID: [392530](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=392530)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/dcf42917c550714d2457947538b9e29d083e872e/subsys/net/lib/ptp/port.c

```
86       return ptp_transport_send(port, msg, idx);
87     }
88    
89     static void port_timer_set_timeout(struct k_timer *timer, uint8_t factor, int8_t log_seconds)
90     {
91       uint64_t timeout = log_seconds < 0 ? (NSEC_PER_SEC * factor) >> (log_seconds * -1) :
>>>     CID 392530:    (OVERFLOW_BEFORE_WIDEN)
>>>     Potentially overflowing expression "1000000000U * factor" with type "unsigned int" (32 bits, unsigned) is evaluated using 32-bit arithmetic, and then used in a context that expects an expression of type "uint64_t" (64 bits, unsigned).
92                                            (NSEC_PER_SEC * factor) << log_seconds;
93    
94       k_timer_start(timer, K_NSEC(timeout), K_NO_WAIT);
95     }
96    
97     static void port_timer_set_timeout_random(struct k_timer *timer,
85    
86       return ptp_transport_send(port, msg, idx);
87     }
88    
89     static void port_timer_set_timeout(struct k_timer *timer, uint8_t factor, int8_t log_seconds)
90     {
>>>     CID 392530:    (OVERFLOW_BEFORE_WIDEN)
>>>     Potentially overflowing expression "1000000000U * factor" with type "unsigned int" (32 bits, unsigned) is evaluated using 32-bit arithmetic, and then used in a context that expects an expression of type "uint64_t" (64 bits, unsigned).
91       uint64_t timeout = log_seconds < 0 ? (NSEC_PER_SEC * factor) >> (log_seconds * -1) :
92                                            (NSEC_PER_SEC * factor) << log_seconds;
93    
94       k_timer_start(timer, K_NSEC(timeout), K_NO_WAIT);
95     }
96    
```

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_OVERFLOW_BEFORE_WIDEN). ([CWE-190](http://cwe.mitre.org/data/definitions/190.html))

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=392530

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

carlescufi, 2024-12-04:
@awojasinski please take another look at Coverity, still complains after your fix.

carlescufi, 2025-01-08:
Fixed by #83158

PR #74945 (2024-06-27): net: lib: ptp: Fix issues raised by static analyzer
Fix minor issues reported by static analysis tool:
- #74799
- #74800 
- #74801
- #74804 
- #74805 
- #74806 
- #74807 
- #74808 
- #74809
- #74811 
- #74812

Fixes #74799
Fixes #74800 
Fixes #74801
Fixes #74804 
Fixes #74805 
Fixes #74806 
Fixes #74807 
Fixes #74808 
Fixes #74809
Fixes #74811 
Fixes #74812
closed 2025-01-08 by carlescufi