zephyrproject-rtos/zephyr #74809: [Coverity CID: 392535] Out-of-bounds write in subsys/net/lib/ptp/port.c

zephyrbot, 2024-06-22:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/dcf42917c550714d2457947538b9e29d083e872e/subsys/net/lib/ptp/port.c#L1019

Category: Memory - corruptions
Function: `ptp_port_init`
Component: Networking
CID: [392535](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=392535)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/dcf42917c550714d2457947538b9e29d083e872e/subsys/net/lib/ptp/port.c#L1019

```
1013                     LOG_WRN("Exceeded number of PTP Ports.");
1014                     return;
1015             }
1016    
1017             port = ports + dds->n_ports;
1018    
>>>     CID 392535:  Memory - corruptions  (OVERRUN)
>>>     Overrunning array of 364 bytes at byte offset 364 by dereferencing pointer "port".
1019             port->iface = iface;
1020             port->best = NULL;
1021             port->socket[PTP_SOCKET_EVENT] = -1;
1022             port->socket[PTP_SOCKET_GENERAL] = -1;
1023    
1024             port->state_machine = dds->time_receiver_only ? ptp_tr_state_machine : ptp_state_machine;
```

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_OVERRUN). ([CWE-119](http://cwe.mitre.org/data/definitions/119.html))

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=392535

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

carlescufi, 2024-12-04:
@awojasinski please take another look at Coverity, still complains after your fix.

carlescufi, 2025-01-08:
Fixed by #83158

PR #74945 (2024-06-27): net: lib: ptp: Fix issues raised by static analyzer
Fix minor issues reported by static analysis tool:
- #74799
- #74800 
- #74801
- #74804 
- #74805 
- #74806 
- #74807 
- #74808 
- #74809
- #74811 
- #74812

Fixes #74799
Fixes #74800 
Fixes #74801
Fixes #74804 
Fixes #74805 
Fixes #74806 
Fixes #74807 
Fixes #74808 
Fixes #74809
Fixes #74811 
Fixes #74812
closed 2025-01-08 by carlescufi